carla-simulator/carla #8290: can you provide more information about how to then load the additional maps.

NayaBaslan, 2024-10-17:
I have a question about a previously close issue.
For loading the additional maps in Carla.

The instructions set, download the zipped package and unzip it inside Import folder and then rerun ./ImportAssets.sh

can you provide more information about how to then load the additional maps.
Usually for the ones already available, i use
`python3 config.py --map Town01`

But the option for additional maps doesnt seem available.
Can you provide me with the right command I need to run to change the map to one of the additional maps?
I am using CARLA 0.9.15 and need to use Town12

_Originally posted by @NayaBaslan in https://github.com/carla-simulator/carla/issues/6062#issuecomment-2419785164_

glopezdiest, 2024-10-18:
Hello @NayaBaslan. The command is the same:

`python3 config.py --map Town12`

The additional maps aren't any different from the normal ones, it's just that they are too big for us to put them in the "main" release, when not everyone will use them.

If the command doesn't work, then it means that the importing didn't finish correctly.

Closing the issue, but feel free to reopen it if needed

closed 2024-10-18 by glopezdiest

NayaBaslan, 2024-10-18:
Thank you. It was indeed that the import did not run properly.