carla-simulator/carla #8291: ValueError: sticky_control: invalid value type

hee-jae-kim, 2024-10-17:
CARLA version: 0.9.15
Platform/OS: ubuntu 22.04

I am running the manual_control_steeringwheel.py and getting an error "ValueError: sticky_control: invalid value type"
Any suggestions about this issue? I guess it's a version compatibility issue. Below is my error

ALSA lib pcm_dmix.c:1032:(snd_pcm_dmix_open) unable to open slave
WARNING: Version mismatch detected: You are trying to connect to a simulator that might be incompatible with this API 
WARNING: Client API version     = 0.9.15-226-g5c2d2978a 
WARNING: Simulator API version  = 0.9.15-221-g4da0c7415 
Traceback (most recent call last):
  File "/home/h2x/carla/PythonAPI/examples/manual_control_steeringwheel_hj.py", line 1403, in <module>
    main()
  File "/home/h2x/carla/PythonAPI/examples/manual_control_steeringwheel_hj.py", line 1395, in main
    game_loop(args)
  File "/home/h2x/carla/PythonAPI/examples/manual_control_steeringwheel_hj.py", line 1114, in game_loop
    world = World(client.get_world(), hud, args.filter)
  File "/home/h2x/carla/PythonAPI/examples/manual_control_steeringwheel_hj.py", line 289, in __init__
    self.restart()
  File "/home/h2x/carla/PythonAPI/examples/manual_control_steeringwheel_hj.py", line 298, in restart
    blueprint_library = self.world.get_blueprint_library()
ValueError: sticky_control: invalid value type

NayaBaslan, 2024-10-18:
what does this command give you?
`dmesg | grep -i joystick`

ytj254, 2024-11-08:
I have a similar error when running run_synchronization.py. My CARLA version is 0.9.15, Platform/OS is ubuntu 22.04. Error message is:
Traceback (most recent call last):
  File "/home/ytj/carla/Co-Simulation/Sumo/run_synchronization.py", line 319, in <module>
    synchronization_loop(arguments)
  File "/home/ytj/carla/Co-Simulation/Sumo/run_synchronization.py", line 238, in synchronization_loop
    carla_simulation = CarlaSimulation(args.carla_host, args.carla_port, args.step_length)
  File "/home/ytj/carla/Co-Simulation/Sumo/sumo_integration/carla_simulation.py", line 34, in __init__
    self.blueprint_library = self.world.get_blueprint_library()
ValueError: sticky_control: invalid value type
Run the command dmesg | grep -i joystick returned:
dmesg: read kernel buffer failed: Operation not permitted

Any suggestions to solve this? Thanks.

totoro52, 2025-01-17:
```
INFO: listening to server 127.0.0.1:2000
INFO:  Found the required file in cache!  Carla/Maps/Nav/Town10HD_Opt.bin 
INFO:  Found the required file in cache!  Carla/Maps/TM/Town10HD_Opt.bin 
Traceback (most recent call last):
  File "D:/carla/WindowsNoEditor/PythonAPI/examples/automatic_control.py", line 884, in <module>
    main()
  File "D:/carla/WindowsNoEditor/PythonAPI/examples/automatic_control.py", line 877, in main
    game_loop(args)
  File "D:/carla/WindowsNoEditor/PythonAPI/examples/automatic_control.py", line 738, in game_loop
    world = World(client.get_world(), hud, args)
  File "D:/carla/WindowsNoEditor/PythonAPI/examples/automatic_control.py", line 136, in __init__
    self.restart(args)
  File "D:/carla/WindowsNoEditor/PythonAPI/examples/automatic_control.py", line 148, in restart
    blueprint_list = get_actor_blueprints(self.world, self._actor_filter, self._actor_generation)
  File "D:/carla/WindowsNoEditor/PythonAPI/examples/automatic_control.py", line 85, in get_actor_blueprints
    bps = world.get_blueprint_library().filter(filter)
ValueError: sticky_control: invalid value type
```

I have the same problem, but I am using windows

wang87325, 2025-05-09:
I have the same problem, how to solve it?

totoro52, 2025-05-09:
> 我也遇到了同样的问题，请问如何解决？

reduced version