carla-simulator/carla #8293: How to add a custom attribute to a vehicle and make it accessible through the python API?

FarahatDeploys, 2024-10-19:
Hello Everyone,

I am working on CARLA server and I am trying to add a new attribute or variable that makes certain behaviour in the vehicle, any idea how can I do so ? is there a tutorial that explains such thing ? 

let's say I have a vehicle X, I would like to introduce a new variable called CurrentMaxVelocity 
and then be able to access it through the python API ?

Thanks in advance !

closed 2024-10-21 by FarahatDeploys

MassaDenis, 2025-01-22:
Hello!

Сould you tell how you solved the problem? Thanks!

FarahatDeploys, 2025-01-22:
Hello Massa, i managed to this by editing the CarlaWheeledVehicle header file and exposed the new variable using Boost library then i did a rebuild for the pythonapi and i was good to go.

i managed to add a new variable called path inside the control object so we managed to have new variable for example control.path and it holds array of the next way point inteded for the vehicle.

let me know if you would like to have more information about how if this fits with you needs

MassaDenis, 2025-01-23:
Hello Farahat! 

I want to add a new attribute to a vehicle, which will contain the license plate as a string variable. This attribute will be set in the BaseVehiclePawn blueprint and must be accessible through the python API. 

If you can provide me with more information for a solution, I would really appreciate it. Thanks!

FarahatDeploys, 2025-01-23:
Hello Massa,

I made this document when i was working on the project, i believe it may help you !

https://nervous-land-34f.notion.site/How-to-Add-New-Variable-for-CARLA-Pipeline-18427a0ded11804498ace5b01ef72562

What i was trying to do in this example is basically add a new variable called augmented reality path and make it accessible from the unreal engine and in the same time expose from the python API
for example  to set the variable for your vehicle through python API you do the following 
control.AugmetnedReality = [] 
control.Throttle = 0.5 
then you apply the contorl 
and from the unreal engine side you can receive this array if you use the get vehicle control and decompose the returned into smaller variables (throttle, steering, ... AugmetnedReality).. etc

I am sorry i am a little busy with other stuff but the approach in the link worked for me hope it helps you 

have a good day

MassaDenis, 2025-01-23:
Thank you so much for your help, have a nice day!