carla-simulator/carla #8295: Build Error in CARLA when Adding new Method to CarlaActor

Zerquer, 2024-10-21:
Build Error in CARLA when Adding SetClientTest Method
CARLA Version: 0.9.12
Platform/OS: Windows 10

Problem Description:
I get a build error when I add even a minimal implementation for ECarlaServerResponse FVehicleActor::SetClientTest(int8_t Test).

Expected Behavior:
The build should complete without errors.

Steps to Reproduce:
Add the following in CarlaActor.h:

```bash
  virtual ECarlaServerResponse SetClientTest(int8_t Test) {
      return ECarlaServerResponse::ActorTypeMismatch;
  }  
  
   virtual ECarlaServerResponse SetClientTest(int8_t Test) final;
```

Add the following in CarlaActor.cpp:

```bash
ECarlaServerResponse FVehicleActor::SetClientTest(int8_t Test)
{
  if (!IsDormant()) {
    auto Vehicle = Cast<ACarlaWheeledVehicle>(GetActor());
    if (Vehicle == nullptr) {
      return ECarlaServerResponse::NotAVehicle;
    }
    Vehicle->SetClientTest(Test);
  }
  return ECarlaServerResponse::Success;
}
```
I have created several methods in the same way, and all of them work fine. However, adding this new method results in a build error, regardless of whether it is a getter or setter method or what the return types are.

I have spent a lot of time debugging and still cannot find the error.

I added the new code in my code files and marked them with `///New Code`. Additionally, I have attached a picture and the output log from my build system

[image]

[2024-10-21_windows.zip](https://github.com/user-attachments/files/17458460/2024-10-21_windows.zip)
[code.zip](https://github.com/user-attachments/files/17458572/code.zip)

Zerquer, 2024-10-22:
Could it be possible that the number of methods used is limited? No matter which method I add, I encounter the mentioned compilation error. However, if I delete some of my previously created functions, it works.

Blyron, 2024-12-03:
You cannot add two methods with the same declaration in C++  you will need to create new declaration.

Blyron, 2024-12-03:
if you want the .cpp code to be effective you should remove 
 virtual ECarlaServerResponse SetClientTest(int8_t Test) {
      return ECarlaServerResponse::ActorTypeMismatch;
  }  
  in the ,h and only let 
  
  virtual ECarlaServerResponse SetClientTest(int8_t Test) final;

closed 2025-05-04 by LuigiFerraioli