carla-simulator/carla #8299: remove traffic lights

CoderXuans, 2024-10-22:
Even though I moved  traffic lights underground and ran the evaluation code for the leaderboard, the traffic lights still appear. I moved the nearby low houses underground, and they disappeared, but when I tried to back up the map data and delete the traffic lights, clicking play caused Unreal Engine to crash.

MislavMores, 2025-01-08:
I am dealing with same problem. Did you solve it?

CoderXuans, 2025-01-09:
No

CoderXuans, 2025-01-17:
@MislavMores 
If you find a good solution, please share it with me.

MislavMores, 2025-01-17:
@CoderXuans 
You can freeze them all in green and then toggle their visibility. In this way, vehicles will behave as if there is no traffic lights and you will not see them on the map. The code is below.

for traffic_light in world.get_actors().filter('traffic.traffic_light'):
        traffic_light.set_state(carla.TrafficLightState.Green)  
        traffic_light.freeze(True)  

traffic_lights = world.get_environment_objects(carla.CityObjectLabel.TrafficLight)
traffic_light_ids = {traffic_light.id for traffic_light in traffic_lights}
world.enable_environment_objects(traffic_light_ids, False)

Another solution is to open map in Unreal Engine and remove traffic lights manually in editor.

CoderXuans, 2025-01-17:
@MislavMores 
I tried opening the map in Unreal Engine and manually deleting the traffic lights in the editor, but after deleting them, Unreal Engine crashes when I click run.
I will try the first method you mentioned, thank you.

MislavMores, 2025-01-17:
No problem.

CoderXuans, 2025-03-13:
@MislavMores 
I found that my CARLA version is relatively low, and there is no `get_environment_objects` function.