carla-simulator/carla #8313: Spawn failed because of collision at spawn position

DarkerSun6, 2024-10-24:


DarkerSun6, 2024-10-24:
CARLA version:carla0.9.14
Platform/OS:ubuntu20.04
Problem you have experienced:
I encountered the following error when placing pedestrians on randomly generated points
Spawn failed because of collision at spawn position
I used the following method to obtain randomly generated points
    walker_position = []
    for j in range(4):
        spawn_1 = world.get_random_location_from_navigation()
        spawn_location = {attr:getattr(spawn_1,attr) for attr in ["x","y","z"]}
        if spawn_location not in walker_position:
            walker_position.append(spawn_location)
I have drawn the error point in this diagram, and it can be seen that there is no collision at that location, as shown in point 2
[image]

What you expected to happen:
I would greatly appreciate it if I could receive your answer

PatrickPromitzer, 2024-11-05:
Hi,
my guess is, that the spawn point is too close to the wall or the floor.
The same problem happens if you spawn 2 pedestrians too close together.
After spawning, you don't have the problem with using way-points (navigation), because it would be enough to be close to the way-point and don't need to be exactly on the location value.

```
get_random_location_from_navigation(self)
This can only be used with walkers. It retrieves a random location to be used as a destination using 
the go_to_location() method in carla.WalkerAIController. This location will be part of a sidewalk. 
Roads, crosswalks and grass zones are excluded. The method does not take into consideration 
locations of existing actors so if a collision happens when trying to spawn an actor, 
it will return an error. Take a look at generate_traffic.py for an example.
```
It looks like the example script doesn't take into account that some navigation points can't be used as a spawn point.

PabloVD, 2024-11-08:
What @PatrickPromitzer comments is the likely reason of the collision. You can avoid possible issues using `try_spawn_actor()` rather than `spawn_actor` ([see here](https://carla.readthedocs.io/en/latest/python_api/#carla.World.try_spawn_actor)), in that way it does not raise an exception if a collision occurs.

closed 2025-04-07 by Blyron