carla-simulator/carla #8319: Windows Build CarlaUE4 VS 2022 - Target framework not supported

code-raccoon, 2024-10-26:
I try to use the Windows Build-from-Source as described in the [official documentation](https://carla.readthedocs.io/en/latest/build_windows/#minor-installations).

**CARLA version**: CARLA 0.9.15
**Platform/OS**: Microsoft Windows 10 Home (10.0.19045 Build 19045)

I tried to follow all windows-build instructions as close as possible, however
- A download for *Visual Studio 2019 Community* is not available anymore, so I used **Visual Studio 2022 Community** (17.11.5)
- As the manual indicates, I selected available Windows 10 and Windows 11 SDKs (as Windows 8.1 SDK is not available)
- "_.NET Framework 4.6.2 development tools_" was not available, so I chose "_.NET Framework 4.6.2 - 4.7.1. development tools_" 

Although the execution of `Setup.bat` and `GenerateProject.bat` proceeded without any issues, I encountered the issue when **opening the UE4.sln**: *Target framework not supported*, which appears the following C# projects 
- GitDependencies
- MemoryProfiler2
- nDisplayLauncher
- nDisplayListener

I already tried some different approaches suggested from [here](https://stackoverflow.com/questions/70022194/open-net-framework-4-5-project-in-vs-2022-is-there-any-workaround) but they didn't work for me. 

Any advice? 

Screenshot: 

[image]

PatrickPromitzer, 2024-11-05:
Hi,
did you try this command?
```
make PythonAPI GENERATOR="Visual Studio 17 2022"
```

Make sure to use the Visual Studio command line.
```
x64 Visual C++ Toolset. In the Workloads section, choose Desktop development with C++. 
This will enable a x64 command prompt that will be used for the build. 
Check that it has been installed correctly by pressing the Windows button and searching for x64. 
Be careful not to open a x86_x64 prompt. 
```

jannisjung, 2025-02-06:
Hi, I assume this is about opening the `UE4.sln` in Visual Studio to build the CARLA UnrealEngine mod.
 
For me the following solution, manually adding the target, did work: (found here: https://thomaslevesque.com/2021/11/12/building-a-project-that-target-net-45-in-visual-studio-2022/)

@code-raccoon On that dialoge I chose 'Download .NET Framework target 4.5 targeting pack (opens browser)' option, to make sure that the affected projects are imported. Unfortunately the necessary pack won't be available)

To add the .NET Framework anyways, do the following steps:  

1. Download the Microsoft.NETFramework.ReferenceAssemblies.net45 from [nuget](https://www.nuget.org/packages/microsoft.netframework.referenceassemblies.net45)
    * find the Download package link on the right side, under the 'About' section.
2. Treat the downloaded file like a zip and extract the contents from `build\.NETFramework\v4.5\`
3. copy the extracted files to: `C:\Program Files (x86)\Reference Assemblies\Microsoft\Framework\.NETFramework\v4.5`
4. Restart Visual Studio

Blyron, 2025-08-05:
The instructions of ue4-dev have been updated to show how to usevs 2022

closed 2025-09-22 by Blyron