carla-simulator/carla #8325: Connecting to opt Map that has unloaded a map layer crashes the Simulation

00Dan00, 2024-10-29:
CARLA version: 0.9.15
Platform/OS: Windows

When unloading map layers and then connecting from another script with a Client immediatly crashes the Simulation on Connection.
`import carla
import random
# Connect to CARLA
client = carla.Client('localhost', 2000)
client.set_timeout(10.0)
world = client.get_world()

world.unload_map_layer(carla.MapLayer.Buildings)

# Get the blueprint library
blueprint_library = world.get_blueprint_library()

# Choose a random vehicle blueprint
vehicle_bp = blueprint_library.filter('vehicle.*')[0]  # e.g., choose the first available vehicle

# Choose a random spawn point
spawn_points = world.get_map().get_spawn_points()
spawn_point = random.choice(spawn_points)  # Randomly select a spawn point

# Spawn the vehicle
vehicle = world.try_spawn_actor(vehicle_bp, spawn_point)

vehicle.set_autopilot(True)`

`import carla
# Connect to CARLA
client = carla.Client('localhost', 2000)
client.set_timeout(10.0)
world = client.get_world()`

Why does this happen and how could I fix it?

00Dan00, 2024-10-29:
Also loading the layer again first doesn't fix it

Fillipko, 2024-12-26:
I'm having the same issue (CARLA version: 0.9.15, OS: Ubuntu 22.04). It looks like after unloading specific map layers (namely `Buildings`, `Ground`, `StreetLights`, and `Props`), calling `client.get_world()` crashes the server. However, unloading the other layers doesn't seem to cause the sever to crash.

marcoCalipari, 2025-01-27:
Hello,

Is there any updates on this? I am facing the same problem with the same carla version

CamiloMartinezM, 2025-04-29:
Same problem