carla-simulator/carla #8343: Trafficlights in Town15

lingyixia, 2024-11-06:
How does the trafficlights generate in Town15? It seemed that they are generated with the startup of the world.

lingyixia, 2024-11-06:
I need to modify these traffic lights，could you help me on how to find where they are？

gsmario, 2024-11-08:
Hello! You're right, all the traffic lights are generated when you start the simulation.
You can look for them with the name "BP_TLOpenDrive"
[image]
You can find a bunch of traffic lights here, for example: 
[image]

And if you need help with the traffic light groups, you can follow this guide: https://carla.readthedocs.io/en/latest/tuto_M_custom_add_tl/

Let me know if I can help you with anything else!

closed 2025-06-13 by Blyron