carla-simulator/carla #8347: make PythonAPI error libmvec.so

Nafyaz, 2024-11-09:
CARLA version: 9.15
Platform/OS: Ubuntu 24.04.1 LTS
Problem you have experienced: Cannot run `make PythoAPI`
What you expected to happen: anything but an error
Steps to reproduce: I was just following this -> https://carla.readthedocs.io/en/latest/build_linux/
Other information (documentation you consulted, workarounds you tried):
I don't know where to start. Been stuck here since yesterday. Went through a ton of previous issues, stackoverflow answers, other clang related package problems. I guess I am going to install ubuntu 22 and try again.

Here is the full stacktrace:
```nafis@slavebook:~/E/carla$ make setup
BuildUE4Plugins.sh: Downloading STREETMAP plugin.
HEAD is now at 260273d Correct descriptor
BuildUE4Plugins.sh: StreetMap Success!
Setup.sh: Retrieving boost.
--2024-11-09 10:52:42--  https://archives.boost.io/release/1.80.0/source/boost_1_80_0.tar.gz
Resolving archives.boost.io (archives.boost.io)... 151.101.67.52, 151.101.195.52, 151.101.3.52, ...
Connecting to archives.boost.io (archives.boost.io)|151.101.67.52|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 136670223 (130M) [application/octet-stream]
Saving to: ‘boost_1_80_0.tar.gz’

boost_1_80_0.tar.gz                           100%[=================================================================================================>] 130.34M   825KB/s    in 3m 6s   

2024-11-09 10:55:50 (716 KB/s) - ‘boost_1_80_0.tar.gz’ saved [136670223/136670223]

Elapsed Time downloading from boost webpage: 188 seconds
Setup.sh: Extracting boost for Python 3.
Elapsed Time Extracting boost for Python: 2 seconds
Building B2 engine..

A C++11 capable compiler is required for building the B2 engine.
Toolset 'clang' does not appear to support C++11.

> clang++ -x c++ -std=c++11  check_cxx11.cpp
/home/nafis/E/UnrealEngine_4.26/Engine/Extras/ThirdPartyNotUE/SDKs/HostLinux/Linux_x64/v17_clang-10.0.1-centos7/x86_64-unknown-linux-gnu/bin/x86_64-unknown-linux-gnu-ld: warning: /usr/lib/gcc/x86_64-linux-gnu/11/../../../x86_64-linux-gnu/crt1.o: unsupported GNU_PROPERTY_TYPE (5) type: 0xc0008002
/home/nafis/E/UnrealEngine_4.26/Engine/Extras/ThirdPartyNotUE/SDKs/HostLinux/Linux_x64/v17_clang-10.0.1-centos7/x86_64-unknown-linux-gnu/bin/x86_64-unknown-linux-gnu-ld: warning: /lib/x86_64-linux-gnu/libm.so.6: unsupported GNU_PROPERTY_TYPE (5) type: 0xc0008002
/home/nafis/E/UnrealEngine_4.26/Engine/Extras/ThirdPartyNotUE/SDKs/HostLinux/Linux_x64/v17_clang-10.0.1-centos7/x86_64-unknown-linux-gnu/bin/x86_64-unknown-linux-gnu-ld: /lib/x86_64-linux-gnu/libm.so.6: unknown type [0x13] section `.relr.dyn'
/home/nafis/E/UnrealEngine_4.26/Engine/Extras/ThirdPartyNotUE/SDKs/HostLinux/Linux_x64/v17_clang-10.0.1-centos7/x86_64-unknown-linux-gnu/bin/x86_64-unknown-linux-gnu-ld: skipping incompatible /lib/x86_64-linux-gnu/libm.so.6 when searching for /lib/x86_64-linux-gnu/libm.so.6
/home/nafis/E/UnrealEngine_4.26/Engine/Extras/ThirdPartyNotUE/SDKs/HostLinux/Linux_x64/v17_clang-10.0.1-centos7/x86_64-unknown-linux-gnu/bin/x86_64-unknown-linux-gnu-ld: cannot find /lib/x86_64-linux-gnu/libm.so.6
/home/nafis/E/UnrealEngine_4.26/Engine/Extras/ThirdPartyNotUE/SDKs/HostLinux/Linux_x64/v17_clang-10.0.1-centos7/x86_64-unknown-linux-gnu/bin/x86_64-unknown-linux-gnu-ld: warning: /lib/x86_64-linux-gnu/libmvec.so.1: unsupported GNU_PROPERTY_TYPE (5) type: 0xc0008002
/home/nafis/E/UnrealEngine_4.26/Engine/Extras/ThirdPartyNotUE/SDKs/HostLinux/Linux_x64/v17_clang-10.0.1-centos7/x86_64-unknown-linux-gnu/bin/x86_64-unknown-linux-gnu-ld: /lib/x86_64-linux-gnu/libmvec.so.1: unknown type [0x13] section `.relr.dyn'
/home/nafis/E/UnrealEngine_4.26/Engine/Extras/ThirdPartyNotUE/SDKs/HostLinux/Linux_x64/v17_clang-10.0.1-centos7/x86_64-unknown-linux-gnu/bin/x86_64-unknown-linux-gnu-ld: skipping incompatible /lib/x86_64-linux-gnu/libmvec.so.1 when searching for /lib/x86_64-linux-gnu/libmvec.so.1
/home/nafis/E/UnrealEngine_4.26/Engine/Extras/ThirdPartyNotUE/SDKs/HostLinux/Linux_x64/v17_clang-10.0.1-centos7/x86_64-unknown-linux-gnu/bin/x86_64-unknown-linux-gnu-ld: cannot find /lib/x86_64-linux-gnu/libmvec.so.1
clang++: error: linker command failed with exit code 1 (use -v to see invocation)

** Note, the C++11 capable compiler is _only_ required for building the B2
** engine. The B2 build system allows for using any C++ level and any other
** supported language and resource in your projects.

You can specify the toolset as the argument, i.e.:
    ./build.sh [options] gcc

Toolsets supported by this script are:
    acc, clang, como, gcc, intel-darwin, intel-linux, kcc, kylix, mipspro,
    pathscale, pgi, qcc, sun, sunpro, tru64cxx, vacpp

For any toolset you can override the path to the compiler with the '--cxx'
option. You can also use additional flags for the compiler with the
'--cxxflags' option.

A special toolset; cxx, is available which is used as a fallback when a more
specific toolset is not found and the cxx command is detected. The 'cxx'
toolset will use the '--cxx' and '--cxxflags' options, if present.

Options:
    --help                  Show this help message.
    --verbose               Show messages about what this script is doing.
    --debug                 Build b2 with debug information, and no
                            optimizations.
    --guess-toolset         Print the toolset we can detect for building.
    --cxx=CXX               The compiler exec to use instead of the detected
                            compiler exec.
    --cxxflags=CXXFLAGS     The compiler flags to use in addition to the
                            flags for the detected compiler.

Failed to build B2 build engine
make: *** [Util/BuildTools/Linux.mk:142: setup] Error 1```

PatrickPromitzer, 2024-11-11:
Hi,
You need clang-8, and Ubuntu 20.04 is supporting it.
With Ubuntu 22.04, you need to edit some files to get it working, because clang-8 and clang-10 is not supported anymore.

The UE5 version of Carla (which should be the next release version) needs Ubuntu 22.04 or you have a similar problem.

Carla 0.9.15.2: Ubuntu 20.04
Carla 0.10.0 (UE5): Ubuntu 22.04

Nafyaz, 2024-11-12:
What are the files I need to edit in Ubuntu 22.04 to get it working?
Also, when is Carla 0.10.0 (UE5) expected to release?

PatrickPromitzer, 2024-11-18:
I can't answer that, because I never did it.
Maybe someone else find this issue who can tell you.

closed 2025-08-05 by Blyron