carla-simulator/carla #8348: failed build step for carla-ros2-native-lib

Nafyaz, 2024-11-09:
Ok, I've uninstalled my Ubuntu 24.04 and installed 22.04.

CARLA version: 9.15
Platform/OS: Ubuntu 22.04.5 LTS
Problem you have experienced: carla-ros2-native-lib build failed
What you expected to happen: properly build package
Steps to reproduce: Following this https://carla.readthedocs.io/en/latest/build_linux_ue5/ (except run 'CarlaSetup.sh' instead of 'bash -x Setup.sh')
Other information (documentation you consulted, workarounds you tried): 
This issue looks similar: https://github.com/nginnever/zogminer/issues/54. So, I tried this (of course replacing 'OpenCL' with 'systemd'.
Here is the last lines of stacktrace:

```
[7/59] Performing configure step for 'carla-ros2-native-lib'
-- Configuring done (0.0s)
-- Generating done (0.0s)
-- Build files have been written to: /home/nafis/D/Softwares/CarlaUE5/Build/Ros2Native/carla-ros2-native-lib-prefix/src/carla-ros2-native-lib-build
[8/59] Performing build step for 'carla-ros2-native-lib'
FAILED: Ros2Native/carla-ros2-native-lib-prefix/src/carla-ros2-native-lib-stamp/carla-ros2-native-lib-build /home/nafis/D/Softwares/CarlaUE5/Build/Ros2Native/carla-ros2-native-lib-prefix/src/carla-ros2-native-lib-stamp/carla-ros2-native-lib-build 
cd /home/nafis/D/Softwares/CarlaUE5/Build/Ros2Native/carla-ros2-native-lib-prefix/src/carla-ros2-native-lib-build && /opt/cmake-3.28.3-linux-x86_64/bin/cmake --build . && /opt/cmake-3.28.3-linux-x86_64/bin/cmake -E touch /home/nafis/D/Softwares/CarlaUE5/Build/Ros2Native/carla-ros2-native-lib-prefix/src/carla-ros2-native-lib-stamp/carla-ros2-native-lib-build
[1/1] Linking CXX shared library libcarla-ros2-native.so
FAILED: libcarla-ros2-native.so 
: && /usr/bin/g++-12 -fPIC -D_GLIBCXX_USE_CXX11_ABI=0   -shared -Wl,-soname,libcarla-ros2-native.so -o libcarla-ros2-native.so CMakeFiles/carla-ros2-native.dir/home/nafis/D/Softwares/CarlaUE5/LibCarla/source/carla/ros2/listeners/BasicListener.cpp.o CMakeFiles/carla-ros2-native.dir/home/nafis/D/Softwares/CarlaUE5/LibCarla/source/carla/ros2/listeners/CarlaListener.cpp.o CMakeFiles/carla-ros2-native.dir/home/nafis/D/Softwares/CarlaUE5/LibCarla/source/carla/ros2/listeners/CarlaSubscriberListener.cpp.o CMakeFiles/carla-ros2-native.dir/home/nafis/D/Softwares/CarlaUE5/LibCarla/source/carla/ros2/publishers/BasicPublisher.cpp.o CMakeFiles/carla-ros2-native.dir/home/nafis/D/Softwares/CarlaUE5/LibCarla/source/carla/ros2/publishers/CarlaClockPublisher.cpp.o CMakeFiles/carla-ros2-native.dir/home/nafis/D/Softwares/CarlaUE5/LibCarla/source/carla/ros2/publishers/CarlaCollisionPublisher.cpp.o CMakeFiles/carla-ros2-native.dir/home/nafis/D/Softwares/CarlaUE5/LibCarla/source/carla/ros2/publishers/CarlaDVSCameraPublisher.cpp.o CMakeFiles/carla-ros2-native.dir/home/nafis/D/Softwares/CarlaUE5/LibCarla/source/carla/ros2/publishers/CarlaDepthCameraPublisher.cpp.o CMakeFiles/carla-ros2-native.dir/home/nafis/D/Softwares/CarlaUE5/LibCarla/source/carla/ros2/publishers/CarlaGNSSPublisher.cpp.o CMakeFiles/carla-ros2-native.dir/home/nafis/D/Softwares/CarlaUE5/LibCarla/source/carla/ros2/publishers/CarlaIMUPublisher.cpp.o CMakeFiles/carla-ros2-native.dir/home/nafis/D/Softwares/CarlaUE5/LibCarla/source/carla/ros2/publishers/CarlaISCameraPublisher.cpp.o CMakeFiles/carla-ros2-native.dir/home/nafis/D/Softwares/CarlaUE5/LibCarla/source/carla/ros2/publishers/CarlaLidarPublisher.cpp.o CMakeFiles/carla-ros2-native.dir/home/nafis/D/Softwares/CarlaUE5/LibCarla/source/carla/ros2/publishers/CarlaLineInvasionPublisher.cpp.o CMakeFiles/carla-ros2-native.dir/home/nafis/D/Softwares/CarlaUE5/LibCarla/source/carla/ros2/publishers/CarlaMapSensorPublisher.cpp.o CMakeFiles/carla-ros2-native.dir/home/nafis/D/Softwares/CarlaUE5/LibCarla/source/carla/ros2/publishers/CarlaNormalsCameraPublisher.cpp.o CMakeFiles/carla-ros2-native.dir/home/nafis/D/Softwares/CarlaUE5/LibCarla/source/carla/ros2/publishers/CarlaOpticalFlowCameraPublisher.cpp.o CMakeFiles/carla-ros2-native.dir/home/nafis/D/Softwares/CarlaUE5/LibCarla/source/carla/ros2/publishers/CarlaRGBCameraPublisher.cpp.o CMakeFiles/carla-ros2-native.dir/home/nafis/D/Softwares/CarlaUE5/LibCarla/source/carla/ros2/publishers/CarlaRadarPublisher.cpp.o CMakeFiles/carla-ros2-native.dir/home/nafis/D/Softwares/CarlaUE5/LibCarla/source/carla/ros2/publishers/CarlaSSCameraPublisher.cpp.o CMakeFiles/carla-ros2-native.dir/home/nafis/D/Softwares/CarlaUE5/LibCarla/source/carla/ros2/publishers/CarlaSemanticLidarPublisher.cpp.o CMakeFiles/carla-ros2-native.dir/home/nafis/D/Softwares/CarlaUE5/LibCarla/source/carla/ros2/publishers/CarlaSpeedometerSensor.cpp.o CMakeFiles/carla-ros2-native.dir/home/nafis/D/Softwares/CarlaUE5/LibCarla/source/carla/ros2/publishers/CarlaTransformPublisher.cpp.o CMakeFiles/carla-ros2-native.dir/home/nafis/D/Softwares/CarlaUE5/LibCarla/source/carla/ros2/subscribers/BasicSubscriber.cpp.o CMakeFiles/carla-ros2-native.dir/home/nafis/D/Softwares/CarlaUE5/LibCarla/source/carla/ros2/subscribers/CarlaEgoVehicleControlSubscriber.cpp.o CMakeFiles/carla-ros2-native.dir/home/nafis/D/Softwares/CarlaUE5/LibCarla/source/carla/ros2/types/CameraInfo.cpp.o CMakeFiles/carla-ros2-native.dir/home/nafis/D/Softwares/CarlaUE5/LibCarla/source/carla/ros2/types/CameraInfoPubSubTypes.cpp.o CMakeFiles/carla-ros2-native.dir/home/nafis/D/Softwares/CarlaUE5/LibCarla/source/carla/ros2/types/CarlaCollisionEvent.cpp.o CMakeFiles/carla-ros2-native.dir/home/nafis/D/Softwares/CarlaUE5/LibCarla/source/carla/ros2/types/CarlaCollisionEventPubSubTypes.cpp.o CMakeFiles/carla-ros2-native.dir/home/nafis/D/Softwares/CarlaUE5/LibCarla/source/carla/ros2/types/CarlaEgoVehicleControl.cpp.o CMakeFiles/carla-ros2-native.dir/home/nafis/D/Softwares/CarlaUE5/LibCarla/source/carla/ros2/types/CarlaEgoVehicleControlPubSubTypes.cpp.o CMakeFiles/carla-ros2-native.dir/home/nafis/D/Softwares/CarlaUE5/LibCarla/source/carla/ros2/types/CarlaLineInvasion.cpp.o CMakeFiles/carla-ros2-native.dir/home/nafis/D/Softwares/CarlaUE5/LibCarla/source/carla/ros2/types/CarlaLineInvasionPubSubTypes.cpp.o CMakeFiles/carla-ros2-native.dir/home/nafis/D/Softwares/CarlaUE5/LibCarla/source/carla/ros2/types/Clock.cpp.o CMakeFiles/carla-ros2-native.dir/home/nafis/D/Softwares/CarlaUE5/LibCarla/source/carla/ros2/types/ClockPubSubTypes.cpp.o CMakeFiles/carla-ros2-native.dir/home/nafis/D/Softwares/CarlaUE5/LibCarla/source/carla/ros2/types/Float32.cpp.o CMakeFiles/carla-ros2-native.dir/home/nafis/D/Softwares/CarlaUE5/LibCarla/source/carla/ros2/types/Float32PubSubTypes.cpp.o CMakeFiles/carla-ros2-native.dir/home/nafis/D/Softwares/CarlaUE5/LibCarla/source/carla/ros2/types/Header.cpp.o CMakeFiles/carla-ros2-native.dir/home/nafis/D/Softwares/CarlaUE5/LibCarla/source/carla/ros2/types/HeaderPubSubTypes.cpp.o CMakeFiles/carla-ros2-native.dir/home/nafis/D/Softwares/CarlaUE5/LibCarla/source/carla/ros2/types/Image.cpp.o CMakeFiles/carla-ros2-native.dir/home/nafis/D/Softwares/CarlaUE5/LibCarla/source/carla/ros2/types/ImagePubSubTypes.cpp.o CMakeFiles/carla-ros2-native.dir/home/nafis/D/Softwares/CarlaUE5/LibCarla/source/carla/ros2/types/Imu.cpp.o CMakeFiles/carla-ros2-native.dir/home/nafis/D/Softwares/CarlaUE5/LibCarla/source/carla/ros2/types/ImuPubSubTypes.cpp.o CMakeFiles/carla-ros2-native.dir/home/nafis/D/Softwares/CarlaUE5/LibCarla/source/carla/ros2/types/NavSatFix.cpp.o CMakeFiles/carla-ros2-native.dir/home/nafis/D/Softwares/CarlaUE5/LibCarla/source/carla/ros2/types/NavSatFixPubSubTypes.cpp.o CMakeFiles/carla-ros2-native.dir/home/nafis/D/Softwares/CarlaUE5/LibCarla/source/carla/ros2/types/NavSatStatus.cpp.o CMakeFiles/carla-ros2-native.dir/home/nafis/D/Softwares/CarlaUE5/LibCarla/source/carla/ros2/types/NavSatStatusPubSubTypes.cpp.o CMakeFiles/carla-ros2-native.dir/home/nafis/D/Softwares/CarlaUE5/LibCarla/source/carla/ros2/types/Odometry.cpp.o CMakeFiles/carla-ros2-native.dir/home/nafis/D/Softwares/CarlaUE5/LibCarla/source/carla/ros2/types/OdometryPubSubTypes.cpp.o CMakeFiles/carla-ros2-native.dir/home/nafis/D/Softwares/CarlaUE5/LibCarla/source/carla/ros2/types/Point.cpp.o CMakeFiles/carla-ros2-native.dir/home/nafis/D/Softwares/CarlaUE5/LibCarla/source/carla/ros2/types/Point32.cpp.o CMakeFiles/carla-ros2-native.dir/home/nafis/D/Softwares/CarlaUE5/LibCarla/source/carla/ros2/types/Point32PubSubTypes.cpp.o CMakeFiles/carla-ros2-native.dir/home/nafis/D/Softwares/CarlaUE5/LibCarla/source/carla/ros2/types/PointCloud2.cpp.o CMakeFiles/carla-ros2-native.dir/home/nafis/D/Softwares/CarlaUE5/LibCarla/source/carla/ros2/types/PointCloud2PubSubTypes.cpp.o CMakeFiles/carla-ros2-native.dir/home/nafis/D/Softwares/CarlaUE5/LibCarla/source [... cut ...] res/CarlaUE5/LibCarla/source/carla/ros2/types/Vector3PubSubTypes.cpp.o  -Wl,-rpath,/home/nafis/D/Softwares/CarlaUE5/Build/Ros2Native/install/lib:  -lSYSTEM  /home/nafis/D/Softwares/CarlaUE5/Build/Ros2Native/install/lib/libfastrtps.so && :
/usr/bin/ld: cannot find -lSYSTEM: No such file or directory
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja: build stopped: subcommand failed.
retry: cmake returned 1, backing off for 10 seconds and trying again...
```

qhaas, 2024-11-11:
Seeing the same error in Ubuntu 22.04 x86-64 when attempting to build commit `ed13b406b23fc28e703780438fdb70ebe6dd56e3` of CARLA branch `ue5-dev` against commit `e1ca23e989bff071e92d22d6a5438c1ac300946a` of the `carla-5.5-experimental` branch of the CARLA UE5 fork, note the UE5 at `$CARLA_UNREAL_ENGINE_PATH` built successfully with `make StandardSet` earlier.
```
$ cmake -G Ninja -S . -B Build \
            --toolchain=$PWD/CMake/LinuxToolchain.cmake \
            -DCMAKE_BUILD_TYPE=Release \
            -DENABLE_ROS2=ON \
            -DBUILD_CARLA_UNREAL=ON \
            -DCARLA_UNREAL_ENGINE_PATH=$CARLA_UNREAL_ENGINE_PATH \
            -DCARLA_UNREAL_BUILD_TYPE=Shipping
...
$ cmake --build Build --config Release --target package
...
[83/83] Linking CXX shared library libcarla-ros2-native.so
FAILED: libcarla-ros2-native.so
: && /usr/bin/g++-12 -fPIC -D_GLIBCXX_USE_CXX11_ABI=0   -shared -Wl,-soname,libcarla-ros2-native.so -o libcarla-ros2-native.so CMakeFiles/carla-ros2-
native.dir/mnt/carla-src/LibCarla/source/carla/ros2/listeners/BasicListener.cpp.o
...
CMakeFiles/carla-ros2-native.dir/mnt/carla-src/LibCarla/source/carla/ros2/types/Vector3.cpp.o CMakeFiles/carla-ros2-native.dir/mnt/carla-src/LibCarla/source/carla/ros2/types/Vector3PubSubTypes.cpp.o  -Wl,-rpath,/mnt/carla-src/Build/Ros2Native/install/lib:  -lSYSTEM  /mnt/carla-src/Build/Ros2Native/install/lib/libfastrtps.so && :
/usr/bin/ld: cannot find -lSYSTEM: No such file or directory
collect2: error: ld returned 1 exit status
```

All packages listed in [`CarlaSetup.sh`](https://github.com/carla-simulator/carla/blob/ue5-dev/CarlaSetup.sh#L54) deployed.

MarcelPiNacy-CVC, 2024-11-29:
Hi @Nafyaz, this was my bad, sorry. The latest commits fix this typo.

closed 2024-11-29 by MarcelPiNacy-CVC