carla-simulator/carla #8352: [BUG] Vehicle Wheel Position

purewater0901, 2024-11-11:
CARLA version: 0.9.15
Platform/OS: Ubuntu
Problem you have experienced: The wheel position values have different scales from the vehicle position. 
front wheel:  Vector3D(x=-16635.164062, y=-9431.089844, z=66.815483)
rear wheel:  Vector3D(x=-16334.700195, y=-9430.637695, z=66.815994)
vehicle position:  Location(x=-164.732162, y=-95.141876, z=0.300000)
What you expected to happen: I assume the wheel position is in cm and vehicle position is in meter. I expect wheel position to be meter as well.
Steps to reproduce: Launch the Carla and place the vehicle anywhere you want.

Blyron, 2024-11-11:
Unreal works in cm, so we cannot do anything in it.

purewater0901, 2024-11-11:
@Blyron Hello Byron, and thank you for your message. So in that case, all I can do is to convert the position of each wheel in meters after I get the value?

Blyron, 2024-11-12:
That's it

closed 2024-11-12 by Blyron

purewater0901, 2024-11-12:
Thank you Blyron!