carla-simulator/carla #8356: camera and lidar data how i synchronize

Vishvamunjani19, 2024-11-12:


Blyron, 2024-11-12:
Please explain more your issue properly

Vishvamunjani19, 2024-11-13:
Hello,

I have problem when I put pedestrian in the map they are running as
randomly in any place.

I am doing multimode traffic predictions so I put 2 camera in 2 traffic
pole and 1 lidar so I have to collect the data from pedestrian, cyclists
and vehicle.

I am also confused about what should I do after collecting the raw data can
you please guide me?

Thank you

Regards,
Vishva

Blyron, 2024-11-13:
No, we cannot guide you on what to do after the data collection but essentialy if you are running in sync mode;

tick()
get_camera_A_data()
get_camera_B_data()
get_lidar_data() 

those 3 data collections above will be syncronized
Those methods does not exist theirselfs but you can create those. 
tick()

Closing the issue as is a question not an issue. Check discussions for this purpouse

closed 2024-11-13 by Blyron

Vishvamunjani19, 2024-11-13:
Hello,

Thank you for information

Can you guide me if I want to put pedestrian person in particular crosswalk
and not in vehicle street so what should I do?

Regards,
Vishva