carla-simulator/carla #8364: Crash when starting server in Editor in UE5.5

PabloVD, 2024-11-13:
There is a crash when pressing play in editor mode in `ue5-dev` branch which occurs often (but not always).

Here is the crash report:
```
LoginId:cdb5cd8a3e3849138ff555c3fccaaf81-000003e8

Assertion failed: Code == CURLM_OK [File:Developer/DerivedDataCache/Private/Http/CurlHttpClient.cpp] [Line: 572] 
Error in curl_multi operation: Unrecoverable error in select/poll

libUnrealEditor-DerivedDataCache.so!UE::FCurlHttpConnectionPool::ThreadLoop() [/home/tda/CARLA/CARLAbuild/UnrealEngine/Engine/Source/Developer/DerivedDataCache/Private/Http/CurlHttpClient.cpp:572]
libUnrealEditor-Core.so!FThreadImpl::Run() [/home/tda/CARLA/CARLAbuild/UnrealEngine/Engine/Source/Runtime/Core/Public/Templates/Function.h:470]
libUnrealEditor-Core.so!FRunnableThreadPThread::Run() [/home/tda/CARLA/CARLAbuild/UnrealEngine/Engine/Source/./Runtime/Core/Private/HAL/PThreadRunnableThread.cpp:25]
libUnrealEditor-Core.so!FRunnableThreadPThread::_ThreadProc(void*) [/home/tda/CARLA/CARLAbuild/UnrealEngine/Engine/Source/Runtime/Core/Private/HAL/PThreadRunnableThread.h:187]
libpthread.so.0!UnknownFunction(0x8608)
libc.so.6!clone(+0x42)

```

Blyron, 2024-11-13:
We should check this;
https://forums.unrealengine.com/t/repeated-crash-on-linux-error-in-curl-multi-operation-unrecoverable-error-in-select-poll/2062133/2

Blyron, 2024-11-26:
https://forums.unrealengine.com/t/unreal-5-4-crash-will-not-start-unreal-zen-storage-server-failed-to-launch/1765413/17