carla-simulator/carla #8367: UE5.5 Bugs with Semantic LIDAR sensor！

uuauto2024, 2024-11-13:
The semantic LiDAR sensor in CARLA UE5 has a bug where it only outputs point cloud points of category 0, as shown in this image. Please tell me how to solve this problem. Thank you!
[image]

where this image is generate by open3d_lidar.py (examples)

PabloVD, 2024-11-14:
Thanks for sharing, we are aware of that problem and we are working in fixing it, hopefully we will get a solution soon.

Blyron, 2024-11-14:
https://github.com/carla-simulator/carla/discussions/8323 As you can see in this discussion, there is a spreedsheet with supported features and ones with issues, We are having issues solving that. So please, wait. We are doing our best. 

Remember we are always open to contributions. Closing this issue as it is known issue

closed 2024-11-14 by Blyron