carla-simulator/carla #8369: GameThread timed out waiting for RenderThread after 60.00 secs

jcuic5, 2024-11-14:
```
travis1@travis-Precision-3660:/mnt/host_travis/projects/carla$ ./CarlaUE4.sh -RenderOffScreen -nosound -fps=10 -carla-rpc-port=2000
4.26.2-0+++UE4+Release-4.26 522 0
Disabling core dumps.
LowLevelFatalError [File:Unknown] [Line: 1214] 
GameThread timed out waiting for RenderThread after 60.00 secs
Signal 11 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554 
CommonUnixCrashHandler: Signal=11
Malloc Size=131160 LargeMemoryPoolOffset=196744 
Malloc Size=131160 LargeMemoryPoolOffset=327928 
Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
Segmentation fault (core dumped)
```
Hey, I meet this problem trying to run carla inside a docker container, which is built with ubuntu 22.04, and the carla version is 0.9.15. Been trying a lot by searching existed information but could not solve it. Appreciate it if there's any suggestion.

chuanye-Wang, 2024-11-16:
same issue，need help

abidnazirisami, 2024-11-19:
Same issue. But instead of running inside a docker container, I am facing the issue while connecting with xrdp. It works find when I do it on the machine without remoting into it. I am using carla 0.9.10.1. My log:

```Shell
4.24.3-0+++UE4+Release-4.24 518 0
Disabling core dumps.
LowLevelFatalError [File:Unknown] [Line: 1155] 
GameThread timed out waiting for RenderThread after 60.00 secs
Signal 11 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554 
CommonUnixCrashHandler: Signal=11
Malloc Size=65535 LargeMemoryPoolOffset=131119 
Malloc Size=173808 LargeMemoryPoolOffset=304944 
Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
CarlaUE4.sh: line 5: 692052 Segmentation fault      (core dumped) "$UE4_PROJECT_ROOT/CarlaUE4/Binaries/Linux/CarlaUE4-Linux-Shipping" CarlaUE4 $@
```

thiagodeborba, 2024-12-11:
Did you guys find any solution??

abidnazirisami, 2024-12-11:
@thiagodeborba I am still facing this issue

tianhao-stan-wu, 2025-02-10:
Same issue here on Ubuntu 22.04 server, can't resolve it.
[image]

JasonShiao, 2025-04-05:
First, try vulkaninfo to check if GPU is detected
If you are using NVIDIA GPU, try install the nvidia proprietary driver (535 works for me but it might be other version for you)
`sudo apt-get install nvidia-driver-535 nvidia-dkms-535`

Kait0, 2025-12-09:
I ran into a similar issue with the NVIDIA driver 580.

Instead of downgrading the driver, what also worked for me was switching from vulkan to the opengl backend with the  `-opengl`.
This only works together with `-RenderOffScreen` as opengl is no longer available when rendering on monitor.