carla-simulator/carla #8380: undefined symbol: typeinfo for AController, AActor,USceneComponent,UObject...

wudi2048, 2024-11-17:
I am encountering a linker error when building CarlaUE5 on Ubuntu 22.04. The error indicates that the type information (typeinfo) for various core Unreal Engine classes, such as AController, AActor, UObject, etc., cannot be found. Below is the error log I receive during the build process:

Errot Log:
```cmake
Building CarlaUnreal...
Determining max actions to execute in parallel (16 physical cores, 24 logical cores)
  Executing up to 16 processes, one per physical core
Using Unreal Build Accelerator local executor to run 2 action(s)
  Storage capacity 40Gb
---- Starting trace ----
UbaSessionServer - Disable remote execution (remote sessions will finish current processes)
------ Building 2 action(s) started ------
[1/2] Link (lld) CarlaUnreal (UBA disabled)
ld.lld: error: undefined symbol: typeinfo for AController
>>> referenced by Module.Carla.12.cpp
>>>               /home/vcc/Desktop/CarlaUE5/Unreal/CarlaUnreal/Intermediate/Build/Linux/x64/CarlaUnreal/Development/Carla/Module.Carla.12.cpp.o:(typeinfo for AWheeledVehicleAIController)
>>> referenced by Module.Carla.8.cpp
>>>               /home/vcc/Desktop/CarlaUE5/Unreal/CarlaUnreal/Intermediate/Build/Linux/x64/CarlaUnreal/Development/Carla/Module.Carla.8.cpp.o:(typeinfo for AWalkerController)
>>> did you mean: vtable for AController
>>> defined in: /home/vcc/Desktop/CarlaUE5/Unreal/CarlaUnreal/Intermediate/Build/Linux/x64/CarlaUnreal/Development/Engine/Module.Engine.37.cpp.o

ld.lld: error: undefined symbol: typeinfo for AActor
>>> referenced by Module.Carla.6.cpp
>>>               /home/vcc/Desktop/CarlaUE5/Unreal/CarlaUnreal/Intermediate/Build/Linux/x64/CarlaUnreal/Development/Carla/Module.Carla.6.cpp.o:(typeinfo for ASensor)
>>> referenced by Module.Carla.6.cpp
>>>               /home/vcc/Desktop/CarlaUE5/Unreal/CarlaUnreal/Intermediate/Build/Linux/x64/CarlaUnreal/Development/Carla/Module.Carla.6.cpp.o:(typeinfo for ASensorSpawnerActor)
>>> referenced by Module.Carla.6.cpp
>>>               /home/vcc/Desktop/CarlaUE5/Unreal/CarlaUnreal/Intermediate/Build/Linux/x64/CarlaUnreal/Development/Carla/Module.Carla.6.cpp.o:(typeinfo for ASkyBase)
>>> referenced 25 more times
>>> did you mean: vtable for AActor
>>> defined in: /home/vcc/Desktop/CarlaUE5/Unreal/CarlaUnreal/Intermediate/Build/Linux/x64/CarlaUnreal/Development/Engine/Module.Engine.25.cpp.o

ld.lld: error: undefined symbol: typeinfo for USceneComponent
>>> referenced by Module.Carla.6.cpp
>>>               /home/vcc/Desktop/CarlaUE5/Unreal/CarlaUnreal/Intermediate/Build/Linux/x64/CarlaUnreal/Development/Carla/Module.Carla.6.cpp.o:(typeinfo for USignComponent)
>>> referenced by Module.Carla.4.cpp
>>>               /home/vcc/Desktop/CarlaUE5/Unreal/CarlaUnreal/Intermediate/Build/Linux/x64/CarlaUnreal/Development/Carla/Module.Carla.4.cpp.o:(typeinfo for UNoWorldOffsetSceneComponent)

ld.lld: error: undefined symbol: typeinfo for UActorComponent
>>> referenced by Module.Carla.6.cpp
>>>               /home/vcc/Desktop/CarlaUE5/Unreal/CarlaUnreal/Intermediate/Build/Linux/x64/CarlaUnreal/Development/Carla/Module.Carla.6.cpp.o:(typeinfo for USpringBasedVegetationComponent)
>>> referenced by Module.Carla.8.cpp
>>>               /home/vcc/Desktop/CarlaUE5/Unreal/CarlaUnreal/Intermediate/Build/Linux/x64/CarlaUnreal/Development/Carla/Module.Carla.8.cpp.o:(typeinfo for UVehicleVelocityControl)
>>> referenced by Module.Carla.3.cpp
>>>               /home/vcc/Desktop/CarlaUE5/Unreal/CarlaUnreal/Intermediate/Build/Linux/x64/CarlaUnreal/Development/Carla/Module.Carla.3.cpp.o:(typeinfo for UCustomTerrainPhysicsComponent)
>>> referenced 1 more times

ld.lld: error: undefined symbol: typeinfo for UObject
>>> referenced by Module.Carla.6.cpp
>>>               /home/vcc/Desktop/CarlaUE5/Unreal/CarlaUnreal/Intermediate/Build/Linux/x64/CarlaUnreal/Development/Carla/Module.Carla.6.cpp.o:(typeinfo for UTaggerDelegate)
>>> referenced by Module.Carla.7.cpp
>>>               /home/vcc/Desktop/CarlaUE5/Unreal/CarlaUnreal/Intermediate/Build/Linux/x64/CarlaUnreal/Development/Carla/Module.Carla.7.cpp.o:(typeinfo for UTrafficLightController)
>>> referenced by Module.Carla.7.cpp
>>>               /home/vcc/Desktop/CarlaUE5/Unreal/CarlaUnreal/Intermediate/Build/Linux/x64/CarlaUnreal/Development/Carla/Module.Carla.7.cpp.o:(typeinfo for UInterface)
>>> referenced 8 more times

ld.lld: error: undefined symbol: typeinfo for UCommandlet
>>> referenced by Module.Carla.4.cpp
>>>               /home/vcc/Desktop/CarlaUE5/Unreal/CarlaUnreal/Intermediate/Build/Linux/x64/CarlaUnreal/Development/Carla/Module.Carla.4.cpp.o:(typeinfo for ULoadAssetMaterialsCommandlet)
>>> referenced by Module.Carla.4.cpp
>>>               /home/vcc/Desktop/CarlaUE5/Unreal/CarlaUnreal/Intermediate/Build/Linux/x64/CarlaUnreal/Development/Carla/Module.Carla.4.cpp.o:(typeinfo for UMoveAssetsCommandlet)
>>> referenced by Module.Carla.4.cpp
>>>               /home/vcc/Desktop/CarlaUE5/Unreal/CarlaUnreal/Intermediate/Build/Linux/x64/CarlaUnreal/Development/Carla/Module.Carla.4.cpp.o:(typeinfo for UPrepareAssetsForCookingCommandlet)

ld.lld: error: undefined symbol: typeinfo for UBlueprintFunctionLibrary
>>> referenced by Module.Carla.4.cpp
>>>               /home/vcc/Desktop/CarlaUE5/Unreal/CarlaUnreal/Intermediate/Build/Linux/x64/CarlaUnreal/Development/Carla/Module.Carla.4.cpp.o:(typeinfo for UMapGenFunctionLibrary)
>>> referenced by Module.Carla.4.cpp
>>>               /home/vcc/Desktop/CarlaUE5/Unreal/CarlaUnreal/Intermediate/Build/Linux/x64/CarlaUnreal/Development/Carla/Module.Carla.4.cpp.o:(typeinfo for UOpenDrive)
>>> referenced by Module.Carla.5.cpp
[... 42 lines cut ...]
>>> referenced by Module.Carla.2.cpp
>>>               /home/vcc/Desktop/CarlaUE5/Unreal/CarlaUnreal/Intermediate/Build/Linux/x64/CarlaUnreal/Development/Carla/Module.Carla.2.cpp.o:(typeinfo for ACarlaHUD)

ld.lld: error: undefined symbol: typeinfo for UWorldSubsystem
>>> referenced by Module.Carla.2.cpp
>>>               /home/vcc/Desktop/CarlaUE5/Unreal/CarlaUnreal/Intermediate/Build/Linux/x64/CarlaUnreal/Development/Carla/Module.Carla.2.cpp.o:(typeinfo for UCarlaLightSubsystem)

ld.lld: error: undefined symbol: typeinfo for APawn
>>> referenced by Module.Carla.3.cpp
>>>               /home/vcc/Desktop/CarlaUE5/Unreal/CarlaUnreal/Intermediate/Build/Linux/x64/CarlaUnreal/Development/Carla/Module.Carla.3.cpp.o:(typeinfo for AEmptyActor)

ld.lld: error: undefined symbol: typeinfo for UMovementComponent
>>> referenced by Module.Carla.1.cpp
>>>               /home/vcc/Desktop/CarlaUE5/Unreal/CarlaUnreal/Intermediate/Build/Linux/x64/CarlaUnreal/Development/Carla/Module.Carla.1.cpp.o:(typeinfo for UBaseCarlaMovementComponent)

ld.lld: error: too many errors emitted, stopping now (use --error-limit=0 to see all errors)
clang++: error: linker command failed with exit code 1 (use -v to see invocation)
Trace file written to /home/vcc/Desktop/UnrealEngine5_carla/Engine/Programs/UnrealBuildTool/Log.uba with size 9.4kb
Total time in Unreal Build Accelerator local executor: 2.74 seconds
Total execution time: 4.20 seconds
FAILED: Unreal/CMakeFiles/carla-unreal /home/vcc/Desktop/CarlaUE5/Build/Unreal/CMakeFiles/carla-unreal 
cd /home/vcc/Desktop/CarlaUE5/Build/Unreal && /home/vcc/Desktop/UnrealEngine5_carla/Engine/Build/BatchFiles/Linux/Build.sh CarlaUnreal Linux Development -project=/home/vcc/Desktop/CarlaUE5/Unreal/CarlaUnreal/CarlaUnreal.uproject -game -buildscw
ninja: build stopped: subcommand failed.
``` 
Has anyone encountered similar issues while building CarlaUE5 on Linux (Ubuntu 22.04)? If so, how was it resolved?

MarcelPiNacy-CVC, 2024-11-22:
Hi,
Try updating to the most recent commits of ue5-dev on CARLA and ue5-dev-carla on the CARLA UE5 fork. You should also delete the build folder. 

Marcel

MarcelPiNacy-CVC, 2024-11-28:
Hi, I'm closing this issue. Feel free to reopen it if anything comes up.

closed 2024-11-28 by MarcelPiNacy-CVC

hrnycze, 2024-12-09:
i have exactly same problem with me own project :( could you give me hint how to fixed it. Thanks!

wudi2048, 2024-12-11:
> i have exactly same problem with me own project :( could you give me hint how to fixed it. Thanks!

Hello, I conducted CarlaSetup.sh again. This script can update files and recompile Carla, and this problem didn't happen.

swwwu, 2025-03-21:
> i have exactly same problem with me own project :( could you give me hint how to fixed it. Thanks!

Hello, Have you solved this problem? Can you give me some suggestion on this issue? Thanks!!