carla-simulator/carla #8390: Make Launch Error

shahbazsiddique69, 2024-11-20:
Parameters.cpp
D:\carla\LibCarla\source\carla\Buffer.h(13,10): error C1083: Cannot open include file: 'boost/asio/buffer.hpp': No such file or directory [D:\carla\Build\libcarla-visualstudio\LibCarla\cmake\client\carla_client_debug.vcxproj]
  (compiling source file '../../../../../LibCarla/source/carla/sensor/s11n/SensorHeaderSerializer.cpp')

  SimpleWaypoint.cpp
D:\carla\LibCarla\source\carla\Time.h(11,10): error C1083: Cannot open include file: 'boost/date_time/posix_time/posix_time_types.hpp': No such file or directory [D:\carla\Build\libcarla-visualstudio\LibCarla\cmake\client\carla_client_debug.vcxproj]
  (compiling source file '../../../../../LibCarla/source/carla/multigpu/primaryCommands.cpp')

  SimulationState.cpp
D:\carla\Build\boost-1.80.0-install\include\boost\enable_shared_from_this.hpp(16,10): error C1083: Cannot open include file: 'boost/smart_ptr/enable_shared_from_this.hpp': No such file or directory [D:\carla\Build\libcarla-visualstudio\LibCarla\cmake\client\carla_client_debug.vcxproj]
  (compiling source file '../../../../../LibCarla/source/carla/trafficmanager/LocalizationUtils.cpp')

  TrackTraffic.cpp
D:\carla\Build\boost-1.80.0-install\include\boost\enable_shared_from_this.hpp(16,10): error C1083: Cannot open include file: 'boost/smart_ptr/enable_shared_from_this.hpp': No such file or directory [D:\carla\Build\libcarla-visualstudio\LibCarla\cmake\client\carla_client_debug.vcxproj]
  (compiling source file '../../../../../LibCarla/source/carla/trafficmanager/MotionPlanStage.cpp')

D:\carla\LibCarla\source\carla\Buffer.h(13,10): error C1083: Cannot open include file: 'boost/asio/buffer.hpp': No such file or directory [D:\carla\Build\libcarla-visualstudio\LibCarla\cmake\client\carla_client_debug.vcxproj]
  (compiling source file '../../../../../LibCarla/source/carla/multigpu/router.cpp')

  TrafficLightStage.cpp
  TrafficManager.cpp
D:\carla\LibCarla\source\carla\Buffer.h(13,10): error C1083: Cannot open include file: 'boost/asio/buffer.hpp': No such file or directory [D:\carla\Build\libcarla-visualstudio\LibCarla\cmake\client\carla_client_debug.vcxproj]
  (compiling source file '../../../../../LibCarla/source/carla/multigpu/secondary.cpp')

  TrafficManagerLocal.cpp
D:\carla\LibCarla\source\carla\Buffer.h(13,10): error C1083: Cannot open include file: 'boost/asio/buffer.hpp': No such file or directory [D:\carla\Build\libcarla-visualstudio\LibCarla\cmake\client\carla_client_debug.vcxproj]
  (compiling source file '../../../../../LibCarla/source/carla/multigpu/secondaryCommands.cpp')

  TrafficManagerRemote.cpp
D:\carla\Build\boost-1.80.0-install\include\boost\enable_shared_from_this.hpp(16,10): error C1083: Cannot open include file: 'boost/smart_ptr/enable_shared_from_this.hpp': No such file or directory [D:\carla\Build\libcarla-visualstudio\LibCarla\cmake\client\carla_client_debug.vcxproj]
  (compiling source file '../../../../../LibCarla/source/carla/trafficmanager/CachedSimpleWaypoint.cpp')

  VehicleLightStage.cpp
D:\carla\LibCarla\source\carla\Buffer.h(13,10): error C1083: Cannot open include file: 'boost/asio/buffer.hpp': No such file or directory [D:\carla\Build\libcarla-visualstudio\LibCarla\cmake\client\carla_client_debug.vcxproj]
  (compiling source file '../../../../../LibCarla/source/carla/trafficmanager/CollisionStage.cpp')

  ROS2.cpp
D:\carla\Build\boost-1.80.0-install\include\boost\enable_shared_from_this.hpp(16,10): error C1083: Cannot open include file: 'boost/smart_ptr/enable_shared_from_this.hpp': No such file or directory [D:\carla\Build\libcarla-visualstudio\LibCarla\cmake\client\carla_client_debug.vcxproj]
  (compiling source file '../../../../../LibCarla/source/carla/trafficmanager/LocalizationStage.cpp')

D:\carla\LibCarla\source\carla\Buffer.h(13,10): error C1083: Cannot open include file: 'boost/asio/buffer.hpp': No such file or directory [D:\carla\Build\libcarla-visualstudio\LibCarla\cmake\client\carla_client_debug.vcxproj]
  (compiling source file '../../../../../LibCarla/source/carla/trafficmanager/SimpleWaypoint.cpp')

D:\carla\Build\boost-1.80.0-install\include\boost\geometry.hpp(17,10): error C1083: Cannot open include file: 'boost/geometry/geometry.hpp': No such file or directory [D:\carla\Build\libcarla-visualstudio\LibCarla\cmake\client\carla_client_debug.vcxproj]
  (compiling source file '../../../../../LibCarla/source/carla/trafficmanager/InMemoryMap.cpp')

D:\carla\Build\boost-1.80.0-install\include\boost\enable_shared_from_this.hpp(16,10): error C1083: Cannot open include file: 'boost/smart_ptr/enable_shared_from_this.hpp': No such file or directory [D:\carla\Build\libcarla-visualstudio\LibCarla\cmake\client\carla_client_debug.vcxproj]
  (compiling source file '../../../../../LibCarla/source/carla/trafficmanager/SimulationState.cpp')

D:\carla\Build\boost-1.80.0-install\include\boost\enable_shared_from_this.hpp(16,10): error C1083: Cannot open include file: 'boost/smart_ptr/enable_shared_from_this.hpp': No such file or directory [D:\carla\Build\libcarla-visualstudio\LibCarla\cmake\client\carla_client_debug.vcxproj]
  (compiling source file '../../../../../LibCarla/source/carla/trafficmanager/Parameters.cpp')

D:\carla\Build\boost-1.80.0-install\include\boost\enable_shared_from_this.hpp(16,10): error C1083: Cannot open include file: 'boost/smart_ptr/enable_shared_from_this.hpp': No such file or directory [D:\carla\Build\libcarla-visualstudio\LibCarla\cmake\client\carla_client_debug.vcxproj]
  (compiling source file '../../../../../LibCarla/source/carla/trafficmanager/TrafficLightStage.cpp')

D:\carla\Build\boost-1.80.0-install\include\boost\enable_shared_from_this.hpp(16,10): error C1083: Cannot open include file: 'boost/smart_ptr/enable_shared_from_this.hpp': No such file or directory [D:\carla\Build\libcarla-visualstudio\LibCarla\cmake\client\carla_client_debug.vcxproj]
  (compiling source file '../../../../../LibCarla/source/carla/trafficmanager/TrackTraffic.cpp')

D:\carla\Build\boost-1.80.0-install\include\boost\enable_shared_from_this.hpp(16,10): error C1083: Cannot open include file: 'boost/smart_ptr/enable_shared_from_this.hpp': No such file or directory [D:\carla\Build\libcarla-visualstudio\LibCarla\cmake\client\carla_client_debug.vcxproj]
  (compiling source file '../../../../../LibCarla/source/carla/trafficmanager/TrafficManagerLocal.cpp')

D:\carla\Build\boost-1.80.0-install\include\boost\enable_shared_from_this.hpp(16,10): error C1083: Cannot open include file: 'boost/smart_ptr/enable_shared_from_this.hpp': No such file or directory [D:\carla\Build\libcarla-visualstudio\LibCarla\cmake\client\carla_client_debug.vcxproj]
  (compiling source file '../../../../../LibCarla/source/carla/trafficmanager/TrafficManagerRemote.cpp')

D:\carla\Build\boost-1.80.0-install\include\boost\enable_shared_from_this.hpp(16,10): error C1083: Cannot open include file: 'boost/smart_ptr/enable_shared_from_this.hpp': No such file or directory [D:\carla\Build\libcarla-visualstudio\LibCarla\cmake\client\carla_client_debug.vcxproj]
  (compiling source file '../../../../../LibCarla/source/carla/trafficmanager/VehicleLightStage.cpp')

D:\carla\LibCarla\source\carla\Buffer.h(13,10): error C1083: Cannot open include file: 'boost/asio/buffer.hpp': No such file or directory [D:\carla\Build\libcarla-visualstudio\LibCarla\cmake\client\carla_client_debug.vcxproj]
  (compiling source file '../../../../../LibCarla/source/carla/ros2/ROS2.cpp')

[... 91 lines cut ...]
  [10/25] Module.Carla.4_of_4.cpp
  D:\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla/Game/CarlaHUD.h(63): warning C4996: 'UWheeledVehicleMovementComponent': PhysX is deprecated. Use the UChaosWheeledVehicleMovementComponent fron the ChaosVehiclePhysics Plugin. Please update your code to the new API before upgrading to the next release, otherwise your project will no longer compile.
  D:\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla/Game/CarlaHUD.h(64): warning C4996: 'UWheeledVehicleMovementComponent': PhysX is deprecated. Use the UChaosWheeledVehicleMovementComponent fron the ChaosVehiclePhysics Plugin. Please update your code to the new API before upgrading to the next release, otherwise your project will no longer compile.
  D:\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla/Actor/ActorAttribute.h(9): fatal error C1083: Cannot open include file: 'compiler/disable-ue4-macros.h': No such file or directory
  [11/25] Module.Carla.gen.6_of_8.cpp
[... cut ...]
  [15/25] Module.CarlaTools.gen.1_of_2.cpp
  D:\carla\Unreal\CarlaUE4\Plugins\CarlaTools\Source\CarlaTools/Public/Commandlet/GenerateTileCommandlet.h(12): fatal error C1083: Cannot open include file: 'compiler/disable-ue4-macros.h': No such file or directory
  [16/25] Module.StreetMapRuntime.cpp
  D:\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla/OpenDrive/OpenDriveGenerator.h(12): fatal error C1083: Cannot open include file: 'compiler/disable-ue4-macros.h': No such file or directory
make: *** [CarlaUE4Editor] Error 6
[image]
****

shahbazsiddique69, 2024-11-20:
Most probably I am mainly facing error with the boost installed. I have download the boost-1.80.0-install file and unzipped it and pasted in the D:\carla\Build\boost-1.80.0-install. Because there was only boost-1.80.0-source file. this solved my boost installation error which was specifically "...skipped 63 targets...
...updated 16474 targets...

    -[install_boost]: [B2 ERROR] An error ocurred while installing using "b2.exe".
    -[install_boost]: Exiting with error..."
    but i think my approach didn't end well and have mistakes. please rectify me with solutions if you have.

lakhanqurban, 2024-12-05:
I am facing the same issue please update if you have found any solution.

lakhanqurban, 2024-12-13:
I had to start from scratch by cloning both repositories (Unreal and Carla) again and you are right it only works with the visual studio 2019.

shahbazsiddique69, 2024-12-13:
Glad it helped

closed 2024-12-27 by shahbazsiddique69