carla-simulator/carla #8391: PythonAPI/util/config.py --osm-path does not load the sidewalk

drblallo, 2024-11-20:
CARLA version:2a2cc5f1beb6219ac7a14a60e1d2c5e728557f4a
Platform/OS: linux

Problem you have experienced: 
config.py --osm-path importer does not import the sidewalk . The crosswalk can be seen, but walkers will still be spawned and follow the crosswalk of the map that existed before the current one. The visual elements of the sidewalk are loaded correctly.

Steps to reproduce: run  PythonAPI/util/config.py --osm-path FILE.osm  followed by PythonAPI/examples/geenrate_traffic.py -w30 -n 0

Other information (documentation you consulted, workarounds you tried): i tried digging into the code, cleaning files, updating to the latest version of everything (except unreal engine itself), but i have found the nature of the issue.

drblallo, 2024-11-27:
i found the root cause of this issue to be the caching mechanism for loading walkers navigation files that is broken. I managed to fix the issue for me by just disabling it. I don't know what is the proper fix.

timkang7, 2024-12-17:
hello mate, could you convert .osm file to .xodr? I am tring to convert it follow Convert OpenStreetMap format to OpenDRIVE format document, but it not work well. Some map cannot convert and some xodr file lost many roads and junctions. BTW, which os do you use, linux or win?

drblallo, 2024-12-17:
on the latest commit i get acceptable results for what regards the conversion alone from openstreetmap to open drive. It is true that some content is lost in the translation.

 I had many issues importing the opendrive file in carla, in the end i modified the source code  and butchered stuff i did not need until it did what i wanted. Not a very upstreammable solution tough.

timkang7, 2024-12-18:
Thank you for your reply! It is true that official document is not work for converting the osm to xodr, I made it success by roadrunner finally.