carla-simulator/carla #8397: Create OpenDrive in CarlaUE4

kuebel329, 2024-11-21:
Hello, I’m a working student and try to use Carla for my first project. I wanted to ask you if you’re happened to know about an error and would be very pleased if you can help me to solve this. I built my own map with CarlaUE4 and when I try to execute python code on this, I get the error “ERROR: unable to parse the OpenDRIVE XML string An error occurred: failed to generate map”. Most likely, this is caused by a missing xodr file because there’s no xodr file for this map in the OpenDrive folder. Unfortunately, I didn’t find a way to export a xodr file for a self-built map. Do you know how to get this done? Thank’s a lot for your help!

Blyron, 2024-11-22:
The Open Drive usually comes along with the map. Open Drive is necesary to have spawn points traffic lights and other stuff. Programs like RoadRunner generates it.  If you have the OpenDrive map, you have to place it next to your umap location inside OpenDriveFolder. Remember that the Opendrive file name must match the umap name. 

For Example 
My umap Path: /Content/Carla/Maps/MyCustomMap/MyMap.umap
My umap Path: /Content/Carla/Maps/MyCustomMap/OpenDrive/MyMap.xodr

We would love to have a OpenDrive files generator but it is not the case. So we cannot help you.

If you do not need autonomus agents like the generate traffic, you can just copy paste one of our xodr files into the proper location and rename it. Remember it will spawn Traffic lights and more things so you would like to edit it to let it empty xodr

ronyshaji, 2024-11-24:
@kuebel329 Hey, I used the following tutorial to import my .xodr and .fbx to CARLA.
It worked for me.

[Add a new Map](https://carla.readthedocs.io/en/latest/tuto_M_add_map_source/)

closed 2025-07-16 by Blyron