carla-simulator/carla #8407: Sync Mode; gets stucked in Loop because clients don't tick

Blyron, 2024-11-25:
When you create a camera sensor in sync mode with res of 4k. If you try to save the image to disk or use it in pygame. The client has unknown behaviour in result of not ticking the server again. 

We need to discover what is going on. It happens when you try to save to disk the image or visualize it through pygame.

PabloVD, 2024-12-02:
Log of the above:
```
[2024.12.02-11.20.04:023][ 95]LogBlueprintUserMessages: [CarlaGameMode_C_2147482473] 0
[2024.12.02-11.21.44:452][175]LogCarla: Sensor 1 '' created in primary server
[2024.12.02-11.21.45:111][182]LogTemp: Loaded OpenDrive file '/home/carla/CarlaUnreal/Content/Carla/Maps/OpenDrive/Town10HD_Opt.xodr'
[2024.12.02-11.21.45:775][189]LogTemp: Loaded OpenDrive file '/home/carla/CarlaUnreal/Content/Carla/Maps/OpenDrive/Town10HD_Opt.xodr'
[2024.12.02-11.21.45:816][189]LogCarla: Spawning actor 'sensor.camera.rgb'
[2024.12.02-11.21.45:816][189]LogCarla: Error: ActorAttribute 'color_saturation' is not a color
[2024.12.02-11.21.45:816][189]LogCarla: Error: ActorAttribute 'highlights_gamma' is not a color
[2024.12.02-11.21.45:816][189]g.TimeoutForBlockOnRenderFence = "300000"
[2024.12.02-11.21.45:816][189]LogCarla: Sensor 2 'sensor.camera.rgb' created in primary server
[2024.12.02-11.21.45:818][189]LogCarla: Spawning actor 'vehicle.ue4.mercedes.ccc'
[2024.12.02-11.21.45:822][189]LogCarla: Door name: SM_MercedesCCC_Door_Front_L
[2024.12.02-11.21.45:822][189]LogCarla: Door name: SM_MercedesCCC_Door_Front_R
[2024.12.02-11.21.45:822][189]LogCarla: Spawning actor 'vehicle.ue4.ford.crown'
[2024.12.02-11.21.45:825][189]LogCarla: Door name: Door_FL
[2024.12.02-11.21.45:825][189]LogCarla: Door name: Door_FR
[2024.12.02-11.21.45:825][189]LogCarla: Door name: Door_RL
[2024.12.02-11.21.45:825][189]LogCarla: Door name: Door_RR
[2024.12.02-11.21.45:826][189]LogCarla: Spawning actor 'vehicle.ue4.mercedes.ccc'
[2024.12.02-11.21.45:828][189]LogCarla: Door name: SM_MercedesCCC_Door_Front_L
[2024.12.02-11.21.45:828][189]LogCarla: Door name: SM_MercedesCCC_Door_Front_R
[2024.12.02-11.21.45:829][189]LogCarla: Spawning actor 'vehicle.lincoln.mkz'
[2024.12.02-11.21.45:831][189]LogCarla: Door name: SM_LincolnDoor_Front_L
[2024.12.02-11.21.45:831][189]LogCarla: Door name: SM_LincolnDoor_Front_R
[2024.12.02-11.21.45:831][189]LogCarla: Door name: SM_LincolnDoor_Back_L
[2024.12.02-11.21.45:831][189]LogCarla: Door name: SM_LincolnDoor_Back_R
[2024.12.02-11.21.45:832][189]LogBlueprintUserMessages: [BP_Lincoln2024_C_2147482212] MaterialNotFound
[2024.12.02-11.21.45:832][189]LogCarla: Spawning actor 'vehicle.ue4.chevrolet.impala'
[2024.12.02-11.21.45:834][189]LogCarla: Spawning actor 'vehicle.ue4.bmw.grantourer'
[2024.12.02-11.21.45:836][189]LogCarla: Spawning actor 'vehicle.ue4.mercedes.ccc'
[2024.12.02-11.21.45:838][189]LogCarla: Door name: SM_MercedesCCC_Door_Front_L
[2024.12.02-11.21.45:838][189]LogCarla: Door name: SM_MercedesCCC_Door_Front_R
[2024.12.02-11.21.45:839][189]LogCarla: Spawning actor 'vehicle.firetruck.actors'
[2024.12.02-11.21.45:842][189]LogCarla: Door name: DoorL
[2024.12.02-11.21.45:842][189]LogCarla: Door name: DoorR
[2024.12.02-11.21.45:843][189]LogCarla: Spawning actor 'vehicle.mini.cooper'
[2024.12.02-11.21.45:846][189]LogCarla: Door name: SM_DoorL_Mini2021
[2024.12.02-11.21.45:846][189]LogCarla: Door name: SM_DoorR_Mini2021
[2024.12.02-11.21.45:847][189]LogCarla: Spawning actor 'vehicle.dodge.charger'
[2024.12.02-11.21.45:850][189]LogCarla: Door name: Door_FL
[2024.12.02-11.21.45:850][189]LogCarla: Door name: Door_FR
[2024.12.02-11.21.45:850][189]LogCarla: Door name: Door_RL
[2024.12.02-11.21.45:850][189]LogCarla: Door name: Door_RR
[2024.12.02-11.21.45:852][189]LogCarla: Spawning actor 'vehicle.mini.cooper'
[2024.12.02-11.21.45:854][189]LogCarla: Door name: SM_DoorL_Mini2021
[2024.12.02-11.21.45:854][189]LogCarla: Door name: SM_DoorR_Mini2021
[2024.12.02-11.21.45:854][189]LogCarla: Spawning actor 'vehicle.ue4.ford.mustang'
[2024.12.02-11.21.45:856][189]LogCarla: Spawning actor 'vehicle.nissan.patrol'
[2024.12.02-11.21.45:860][189]LogCarla: Door name: SM_DoorL_Patrol2021
[2024.12.02-11.21.45:860][189]LogCarla: Door name: SM_DoorR_Patrol2021
[2024.12.02-11.21.45:860][189]LogCarla: Door name: SM_DoorRL_Patrol2021
[2024.12.02-11.21.45:860][189]LogCarla: Door name: SM_DoorRR_Patrol2021
[2024.12.02-11.21.45:861][189]LogCarla: Spawning actor 'vehicle.taxi.ford'
[2024.12.02-11.21.45:864][189]LogCarla: Door name: Door_FL
[2024.12.02-11.21.45:864][189]LogCarla: Door name: Door_FR
[2024.12.02-11.21.45:864][189]LogCarla: Door name: Door_RL
[2024.12.02-11.21.45:864][189]LogCarla: Door name: Door_RR
[2024.12.02-11.21.45:865][189]LogCarla: Spawning actor 'vehicle.dodgecop.charger'
[2024.12.02-11.21.45:869][189]LogCarla: Door name: Door_FL
[2024.12.02-11.21.45:869][189]LogCarla: Door name: Door_FR
[2024.12.02-11.21.45:869][189]LogCarla: Door name: Door_RL
[2024.12.02-11.21.45:869][189]LogCarla: Door name: Door_RR
[2024.12.02-11.21.45:871][189]LogBlueprintUserMessages: [BP_ChargerCop2024_C_2147482063] MaterialNotFound
[2024.12.02-11.21.45:871][189]LogBlueprintUserMessages: [BP_ChargerCop2024_C_2147482063] MaterialNotFound
[2024.12.02-11.21.45:871][189]LogCarla: Spawning actor 'vehicle.ue4.ford.mustang'
[2024.12.02-11.21.45:873][189]LogCarla: Spawning actor 'vehicle.ue4.ford.mustang'
[2024.12.02-11.21.45:875][189]LogCarla: Spawning actor 'vehicle.ue4.chevrolet.impala'
[2024.12.02-11.21.45:877][189]LogCarla: Spawning actor 'vehicle.taxi.ford'
[2024.12.02-11.21.45:880][189]LogCarla: Door name: Door_FL
[... 81 lines cut ...]
[2024.12.02-11.21.49:144][189]LogOutputDevice: Error: 
[2024.12.02-11.21.49:154][189]LogStats:                SubmitErrorReport -  0.000 s
[2024.12.02-11.21.49:170][189]LogStats:                    SendNewReport -  0.017 s
[... cut ...]
[2024.12.02-11.21.50:496][191]LogCarla: Spawning actor 'controller.ai.walker'
[2024.12.02-11.21.50:497][191]LogCarla: Spawning actor 'controller.ai.walker'
[2024.12.02-11.21.50:497][191]LogCarla: Spawning actor 'controller.ai.walker'
[2024.12.02-11.21.50:497][191]LogCarla: Spawning actor 'controller.ai.walker'
[2024.12.02-11.21.50:497][191]LogCarla: Spawning actor 'controller.ai.walker'
[2024.12.02-11.21.50:497][191]LogCarla: Spawning actor 'controller.ai.walker'
[2024.12.02-11.21.50:497][191]LogCarla: Spawning actor 'controller.ai.walker'
```

PabloVD, 2024-12-02:
This problem does not appear if:

- the same script runs in asynchronous mode (with the `asynch` flag)
- the camera is of lower resolution (e.g. 1080p)

PabloVD, 2024-12-02:
Some updates:

- Removing the `camera.listen` line in the snippet above the script works as expected and the server does not get stuck. Thus, the problem may be related to the library saving images to disk or moving them to the pygame screen, which in principle are boost::gil and libpng.
- I could not reproduce the issue with a manual control in synch with 4k resolution (in neither editor nor a package).
- Note that this is not a crash, the server just gets stuck. For instance, if afterwards we set it in asynchronous with other script, the server keeps running as usual:

```python
settings = world.get_settings()
settings.synchronous_mode = False
world.apply_settings(settings)
```

PabloVD, 2024-12-05:
Issue no reproducible in all systems. It can be fixed increasing the timeout in `client.set_timeout(1000.0)`.

closed 2024-12-05 by PabloVD