carla-simulator/carla #8419: Semantic Lidar raw_data is wrong

glopezdiest, 2024-11-27:
Each data point in the semantic LIDAR has 4 values (_point_, _cos_inc_angle_, _object_idx_ and _object_tag_). These values are fine if extracted using the SemanticLidarDetection class, but if the measurement is extracted using its _raw_data_, the last two elements, which correspong to  _object_idx_ and _object_tag_ return all 0 values.

glopezdiest, 2024-11-27:
Attaching the script to test it

[lidar_test.zip](https://github.com/user-attachments/files/17933652/lidar_test.zip)

PatrickPromitzer, 2024-11-28:
Hi,
the reason for your code giving out 0 values is, that you convert Integer into float.
```
Array containing the point cloud with instance and semantic information. For each point, four 32-bits floats are stored.
XYZ coordinates.
cosine of the incident angle.
Unsigned int containing the index of the object hit.
Unsigned int containing the semantic tag of the object it.
```
- x (float32)
- y (float32)
- z (float32)
- angle (float32)
- object index (int32)
- object tag (int32)

glopezdiest, 2024-11-28:
Right, that's correct! Thanks you

closed 2024-11-28 by glopezdiest

oskarnatan, 2025-04-11:
Hi, is there a way to convert object tag (int32) into class ID?
I am getting this data when I print the object tag

[1.4e-45 1.4e-45 1.4e-45 ... 1.1e-44 1.1e-44 1.1e-44] [0.0e+00 1.4e-45 2.8e-45 4.2e-45 7.0e-45 9.8e-45 1.1e-44 1.3e-44 1.4e-44 1.5e-44 2.1e-44 2.5e-44 2.7e-44 2.8e-44 2.9e-44]

PatrickPromitzer, 2025-04-11:
> Hi, is there a way to convert object tag (int32) into class ID? I am getting this data when I print the object tag
> 
> [1.4e-45 1.4e-45 1.4e-45 ... 1.1e-44 1.1e-44 1.1e-44] [0.0e+00 1.4e-45 2.8e-45 4.2e-45 7.0e-45 9.8e-45 1.1e-44 1.3e-44 1.4e-44 1.5e-44 2.1e-44 2.5e-44 2.7e-44 2.8e-44 2.9e-44]

Hi,
the tags has to be full numbers (Integer) and not floats.
https://carla.readthedocs.io/en/0.9.15/ref_sensors/#semantic-segmentation-camera