carla-simulator/carla #8427: CarlaUE5.5 Window11 Building Error

timmao955, 2024-12-01:
Hi, I'm building carla-ue5.5 on Windows11 and have some problems for downing the dependencies of carla. The log is as follows: 

PS D:\carla\CarlaUE5> cmake -G Ninja -S . -B Build -DCMAKE_BUILD_TYPE=Release -DBUILD_CARLA_UNREAL=ON -DCARLA_UNREAL_ENGINE_PATH=%CARLA_UNREAL_ENGINE_PATH%
-- The C compiler identification is MSVC 19.42.34435.0
-- The CXX compiler identification is MSVC 19.42.34435.0
-- The ASM compiler identification is MSVC
-- Found assembler: C:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.42.34433/bin/Hostx86/x86/cl.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.42.34433/bin/Hostx86/x86/cl.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.42.34433/bin/Hostx86/x86/cl.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- CARLA: Using D:/carla/UnrealEngine5_carla as Unreal Engine root path.
-- Performing Test HAS_MSVC_UTF8
-- Performing Test HAS_MSVC_UTF8 - Success
-- CARLA: Checking for /EHsc support
-- Performing Test HAS_EXCEPTIONS_FLAG
-- Performing Test HAS_EXCEPTIONS_FLAG - Success
-- CARLA: Checking for /GR support
-- Performing Test HAS_RTTI_FLAG
-- Performing Test HAS_RTTI_FLAG - Success
-- Performing Test HAS_macro_redefined
-- Performing Test HAS_macro_redefined - Failed
-- Performing Test HAS__4005
-- Performing Test HAS__4005 - Success
-- Performing Test HAS_incompatible_pointer_types
-- Performing Test HAS_incompatible_pointer_types - Failed
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - not found
-- Found Threads: TRUE
-- CARLA: Downloading sqlite3...
-- CARLA: Downloading zlib...
-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Looking for stdint.h
-- Looking for stdint.h - found
-- Looking for stddef.h
-- Looking for stddef.h - found
-- Check size of off64_t
-- Check size of off64_t - failed
-- Looking for fseeko
-- Looking for fseeko - not found
-- Looking for unistd.h
-- Looking for unistd.h - not found
-- CARLA: Downloading libpng...
CMake Deprecation Warning at Build/_deps/libpng-src/CMakeLists.txt:33 (cmake_minimum_required):
  Compatibility with CMake < 3.10 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

CMake Deprecation Warning at Build/_deps/libpng-src/CMakeLists.txt:34 (cmake_policy):
  Compatibility with CMake < 3.10 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- Performing Test HAVE_LD_VERSION_SCRIPT
-- Performing Test HAVE_LD_VERSION_SCRIPT - Failed
-- Performing Test HAVE_SOLARIS_LD_VERSION_SCRIPT
-- Performing Test HAVE_SOLARIS_LD_VERSION_SCRIPT - Failed
-- CARLA: Downloading boost...
-- CARLA: Downloading eigen...
-- CARLA: Downloading rpclib...
-- CARLA: Downloading recastnavigation...`

################################
After a very long time, I have received this: 

[1390 bytes data]

  [4041 bytes data]

  HTTP/2 stream 1 was not closed cleanly: PROTOCOL_ERROR (err 1)

  Connection #1 to host objects.githubusercontent.com left intact

          --- LOG END ---

FAILED: boost-populate-prefix/src/boost-populate-stamp/boost-populate-download D:/carla/CarlaUE5/Build/_deps/boost-subbuild/boost-populate-prefix/src/boost-populate-stamp/boost-populate-download
C:\WINDOWS\system32\cmd.exe /C "cd /D D:\carla\CarlaUE5\Build\_deps && D:\cmake\bin\cmake.exe -DCMAKE_MESSAGE_LOG_LEVEL=VERBOSE -P D:/carla/CarlaUE5/Build/_deps/boost-subbuild/boost-populate-prefix/src/boost-populate-stamp/download-boost-populate.cmake && D:\cmake\bin\cmake.exe -DCMAKE_MESSAGE_LOG_LEVEL=VERBOSE -P D:/carla/CarlaUE5/Build/_deps/boost-subbuild/boost-populate-prefix/src/boost-populate-stamp/verify-boost-populate.cmake && D:\cmake\bin\cmake.exe -DCMAKE_MESSAGE_LOG_LEVEL=VERBOSE -P D:/carla/CarlaUE5/Build/_deps/boost-subbuild/boost-populate-prefix/src/boost-populate-stamp/extract-boost-populate.cmake && D:\cmake\bin\cmake.exe -E touch D:/carla/CarlaUE5/Build/_deps/boost-subbuild/boost-populate-prefix/src/boost-populate-stamp/boost-populate-download"
ninja: build stopped: subcommand failed.

CMake Error at D:/cmake/share/cmake-3.31/Modules/FetchContent.cmake:1918 (message):
  Build step for boost failed: 1
Call Stack (most recent call first):
  D:/cmake/share/cmake-3.31/Modules/FetchContent.cmake:1609 (__FetchContent_populateSubbuild)
  D:/cmake/share/cmake-3.31/Modules/FetchContent.cmake:2145:EVAL:2 (__FetchContent_doPopulation)
  D:/cmake/share/cmake-3.31/Modules/FetchContent.cmake:2145 (cmake_language)
  D:/cmake/share/cmake-3.31/Modules/FetchContent.cmake:2384 (__FetchContent_Populate)
  CMake/Dependencies.cmake:50 (FetchContent_MakeAvailable)
  CMake/Dependencies.cmake:277 (carla_dependencies_make_available)
  CMakeLists.txt:85 (include)

-- Configuring incomplete, errors occurred!
PS D:\carla\CarlaUE5>

timmao955, 2024-12-01:
And I followed #8039 , but still stuck for the same log outputs as above.

MarcelPiNacy-CVC, 2024-12-02:
Hi, looks like a download problem. For now you can try deleting the Build folder and passing the option `-DPREFER_CLONE=ON`.

closed 2024-12-08 by MarcelPiNacy-CVC

Miayui, 2025-03-18:
Hi, could you tell me where can I set `DPREFER_CLONE=ON`? I didn't find this option in the .txt or .cmake files.