carla-simulator/carla #8431: LIDAR sensor coordinate system

johnren-code, 2024-12-03:
I'd like to know what coordinate system the acquired LiDAR coordinates are relative to, and what is the relative position of this coordinate system to the orientation of the vehicle? Is there a detailed schematic to explain this?

PatrickPromitzer, 2024-12-03:
Hi,
the Lidar coordinates are in the coordinate system of the Lidar.

If you don't attach the Lidar to an other Actor (for example a vehicle), you have the global coordinate system in which the lidar is spawned, and a sub coordinate system of the Lidar.

Example:
Lidar spawn point 
xyz = 0.0, 0.0, 0.0
roll, pitch, yaw = 0.0, 0.0, 0.0
global coordinate system = Lidar coordinate system = Lidar pints are in the global coordinate system

Example 2:
Lidar spawn point 
xyz = 0.0, 0.0, 1.0
roll, pitch, yaw = 0.0, 0.0, 0.0
The Lidar coordinate system has an offset of z=1.0, and the points of the Lidar have an offset of z=1.0
If you add -1.0 to the Z values of the Lidar points, you would convert the pont into the global coordinate system.
If roll, pitch, and/or yaw is not 0.0 for the spawn point, you need to take that into account, too.

Example 3:
Lidar is attached to a Car
Car xyz = 1.0, 0.0, 0.0 (offset from global coordinate system)
Lidar xyz = 0.0, 0.0, 1.0 (offset from car coordinate system)
Now you have an global, car, and Lidar coordinate system.
- global
    - car
        - Lidar

You would have to convert the Lidar points into the car coordinate system, and then in the global coordinate system.
- global coordinate system: Lidar Point xyz = 1.0, 0.0, 1.0
    - car coordinate system:  Lidar Point xyz = 0.0, 0.0, 1.0
        - Lidar coordinate system: Lidar Point xyz = 0.0, 0.0, 0.0

With roll, pitch, and yaw it will be more complicated to convert, but are the same steps.
I hope this answers your question.

PatrickPromitzer, 2024-12-03:
This will not work.

If the pitch and yaw of your vehicle changes, you will have an offset of your Lidar x and y because of the rotation.
[image]

johnren-code, 2024-12-03:
does the lidar coordinate system look like this? (x-> ahead, y->left, z->up)

PatrickPromitzer, 2024-12-03:
Carla is using a left handed coordinate system.

The coordinate system looks like this.
[image]

johnren-code, 2024-12-04:
OK, I understand, thank you very much for your reply! Actually I use the semantic lidar from carla-ros-bridge. I want to get the lidar coordinate-> world coordiate transform at per frame, now I use the example code from carla-ros-bridge to get the transform : `transform = tf2::transformToEigen (tf_buffer_.lookupTransform(fixed_frame_, cloud->header.frame_id,  pcl_conversions::fromPCL (cloud->header.stamp), ros::Duration(1)));`. But it requires the lidar msg, now I just want to get the lidar coordinate to world coordinate transform without the lidar msg, is there any other solutions?

PatrickPromitzer, 2024-12-04:
I only worked with the Python part of Carla and never used the carla-ros-bride before.

In Python, you can use the actor instance to convert the transform values or create a Transform instance yourself.

```python
actor_transform = actor.get_transform()
new_location = carla.Location(
    x=0.0,
    y=0.0,
    z=1.0
)
actor_transform.transform(new_location)
new_x, mew_y, new_z = new_location.x, new_location.y, new_location.z

# create your own carla Transform instance
carla_transform = carla.Transform(
  carla.Location(
    x=0.0,
    y=0.0,
    z=3.0),
  carla.Rotation(
    pitch=0.0,
    yaw=0.0,
    roll=0.0)
)
```
You could test if something similar works in your code.

Another way would be to convert it yourself, but for that you need to look up how to calculate it.

johnren-code, 2024-12-04:
OK, Thank you very much! It helps me a lot!