carla-simulator/carla #8450: Instance Segmentation camera with CityScapes palette same than semantic segmentation

PabloVD, 2024-12-09:
In the `ue5-dev` branch, the instance segmentation camera with the CityScapes palette provides the same image than the semantic segmentation sensor with that palette. However, the instance sensor in raw seem to provide the correct output, different hues for each actor. Hence,there is a likely issue with the CityScapes palette for instance segmentation.

glopezdiest, 2024-12-10:
This is actually not an issue as it doesn't make sense for the instance segmentation camera to be converter into CityScapesPalette

PabloVD, 2024-12-10:
Removed CityScapesPalette for instance segmentation cameras with [this PR](https://github.com/carla-simulator/carla/pull/8456).

closed 2024-12-10 by PabloVD