carla-simulator/carla #8466: Recent content prevents to create package in Ubuntu for UE5.5

PabloVD, 2024-12-11:
In `ue5-dev` branch, with recent changes in content (commit [eb6525d](https://bitbucket.org/carla-simulator/carla-content/commits/eb6525d2f9a65ae620492dce578216d1e779a5fe) in `ue5-dev` content branch) I was unable to create a package.

Moving to a previous commit [ae404b8](https://bitbucket.org/carla-simulator/carla-content/commits/ae404b8aefaf45fa94126b830c1dd66afa0e3da8) the package was created successfully. Therefore, there is something in the new content changes which conflicts with the package creation procedure.

PabloVD, 2024-12-11:
Log:

```
LogCook: Display: SkipOnlyEditorOnly is enabled, unsolicited packages will not be cooked unless they are referenced from the cooked version of the instigator package.
LogAssetRegistry: Display: Package /home/tda/CARLA/CARLAbuild/carlaUE5/Unreal/CarlaUnreal/Content/Carla/HoudiniEngine/DT_Veg_Asset.uasset uses an unknown custom version and cannot be loaded for the AssetRegistry
LogAssetRegistry: Display: Package /home/tda/CARLA/CARLAbuild/carlaUE5/Unreal/CarlaUnreal/Content/Carla/Static/Procedural/PCG/PCG_Boulder_01.uasset uses an unknown custom version and cannot be loaded for the AssetRegistry
LogAssetRegistry: Display: Package /home/tda/CARLA/CARLAbuild/carlaUE5/Unreal/CarlaUnreal/Content/Carla/HoudiniEngine/DT_BuildingLoader.uasset uses an unknown custom version and cannot be loaded for the AssetRegistry
LogAssetRegistry: Display: Package /home/tda/CARLA/CARLAbuild/carlaUE5/Unreal/CarlaUnreal/Content/Carla/HoudiniEngine/DT_BuildingLoader_2.uasset uses an unknown custom version and cannot be loaded for the AssetRegistry
LogAssetRegistry: Display: Triggering cache save on discovery complete
LogEditorDomain: Display: Cycle detected in parents of class /Script/RigVMDeveloper.RigVMUnitNode. Allow flags for editordomain and iterative cooking will be invalid.
LogEditorDomain: Display: Cycle detected in parents of class /Script/LiveLinkGraphNode.K2Node_EvaluateLiveLinkFrameWithSpecificRole. Allow flags for editordomain and iterative cooking will be invalid.
LogZenStore: Display: Establishing oplog 'CarlaUnreal.6193375e/EditorDomain'
LogZenStore: Display: Zen project 'CarlaUnreal.6193375e' already exists
LogZenStore: Display: Zen oplog 'CarlaUnreal.6193375e/EditorDomain' already exists
LogCook: Display: Waiting for Asset Registry
LogAssetRegistry: Display: Asset registry cache written as 109.2 MiB to ../../../../carlaUE5/Unreal/CarlaUnreal/Intermediate/CachedAssetRegistry_*.bin
LogCook: Display: Cook invalidated for platform Linux because CookSettings file /home/tda/CARLA/CARLAbuild/carlaUE5/Unreal/CarlaUnreal/Saved/Cooked/Linux/CarlaUnreal/Metadata/CookedSettings.txt is invalid. Clearing previously cooked packages.
LogCook: Display: FULL COOK: -iterative was specified, but global settings have changed and all previously cook packages are invalidated. Deleting previously cooked packages for platform Linux and recooking all packages discovered in the current cook.
LogCook: Display: Sandbox cleanup took 0.048 seconds for platforms Linux
LogCook: Display: Compiling global changed shaders for platform 'Linux'
Failed to find game directory: /home/tda/CARLA/CARLAbuild/UnrealEngine/ShaderCompileWorker/Binaries
LogCook: Warning: Unable to find package for cooking /Game/Carla/Static/Dynamic/00_Opt/SM_Barrel_Opt. Instigator: { StartupSoftObjectPath }.
LogCook: Warning: Unable to find package for cooking /Game/Carla/Static/Dynamic/00_Opt/SM_PlasticTable_Opt. Instigator: { StartupSoftObjectPath }.
LogCook: Display: Discovering localized assets for cultures: en
LogCook: Display: Found 0 localized assets
LogShaderCompilers: Display: ================================================
LogShaderCompilers: Display: === FShaderJobCache stats ===
LogShaderCompilers: Display: Total job queries 16, among them cache hits 0 (0.00%), DDC hits 15 (93.75%), Duplicates 0 (0.00%)
LogShaderCompilers: Display: Tracking 16 distinct input hashes that result in 15 distinct outputs (93.75%)
LogShaderCompilers: Display: RAM used: 67,84 KiB of 3,15 GiB budget. Usage: 0.00%
LogShaderCompilers: Display: === Shader Compilation stats ===
LogShaderCompilers: Display: Shaders Compiled: 1
LogShaderCompilers: Display: Jobs assigned 1, completed 1 (100%)
LogShaderCompilers: Display: Time job spent in pending queue: average 0.07 s, longest 0.07 s
LogShaderCompilers: Display: Job execution time: average 0.40 s, max 0.40 s
LogShaderCompilers: Display: Job life time (pending + execution): average 0.47 s, max 0.47
LogShaderCompilers: Display: Shader code size: average 3,991 KiB, min 3,991 KiB, max 3,991 KiB
LogShaderCompilers: Display: Time at least one job was in flight (either pending or executed): 0.47 s
LogShaderCompilers: Display: Mutex wait stall in FShaderJobCache::SubmitJobs:  0.00%
LogShaderCompilers: Display: Jobs were issued in 1 batches (only local compilation was used), average 1.00 jobs/batch
LogShaderCompilers: Display: Average processing rate: 2.11 jobs/sec
LogShaderCompilers: Display: Total thread time: 0,078 s
LogShaderCompilers: Display: Total thread preprocess time: 0,164 s
LogShaderCompilers: Display: Percentage time preprocessing: 208.73%
LogShaderCompilers: Display: Effective parallelization: 0.17 (times faster than compiling all shaders on one thread). Compare with number of workers: 13
LogShaderCompilers: Display: Top 5 most expensive shader types by average time:
LogShaderCompilers: Display:                                      FEmitSegmentationInfoPS (compiled    1 times, average 0.08 sec, max 0.08 sec, min 0.08 sec)
LogShaderCompilers: Display:                                    FEmitCustomDepthStencilPS (compiled    0 times, average 0.00 sec, max 0.00 sec, min 0.00 sec)
LogShaderCompilers: Display:                                            FEmitSceneDepthPS (compiled    0 times, average 0.00 sec, max 0.00 sec, min 0.00 sec)
LogShaderCompilers: Display:                                            FEmitHitProxyIdPS (compiled    0 times, average 0.00 sec, max 0.00 sec, min 0.00 sec)
[... cut ...] oject=/home/tda/CARLA/CARLAbuild/carlaUE5/Unreal/CarlaUnreal/CarlaUnreal.uproject -nocompileeditor -nop4 -cook -stage -archive -package -iterate -clientconfig=Shipping -TargetPlatform=Linux -Platform=Linux -prereqs -build -stagingdirectory=/home/tda/CARLA/CARLAbuild/carlaUE5/Build/Linux-Release/Package/StagedBuilds/Carla-0.10.0-Linux-Shipping -archivedirectory=/home/tda/CARLA/CARLAbuild/carlaUE5/Build/Linux-Release/Package/Carla-0.10.0-Linux-Shipping
ninja: build stopped: subcommand failed.

```

Blyron, 2024-12-11:
I do not get this issue

Blyron, 2024-12-13:
This has been fixed by removing nanite warnings

closed 2024-12-13 by Blyron

qhaas, 2025-05-14:
FYI, seeing this issue in the tagged 0.10.0 release in Ubuntu 22.04:
* carla: commit ada75f920642e18cace0a9f85ecf9d4077ddb531 tag: 0.10.0
* carla-content: commit 518f45bd010c0dea7239b55eed27d01ba132934e tag: 0.10.0
* carla-ue5 fork: commit 6cf20383302899026a5e777219ba4206afe47d38 tag: 0.10.0

```
$ cmake --build Build --target package
...
LoadErrors: Warning: CreateExport: Failed to load Outer for resource 'DefaultSceneRoot': BP_Sky_C /Game/Carla/Maps/OpenDriveMap.OpenDriveMap:PersistentLevel.BP_Sky_2
LoadErrors: Warning: CreateExport: Failed to load Outer for resource 'ExponentialHeightFog': BP_Sky_C /Game/Carla/Maps/OpenDriveMap.OpenDriveMap:PersistentLevel.BP_Sky_2
LoadErrors: Warning: CreateExport: Failed to load Outer for resource 'Planet': BP_Sky_C /Game/Carla/Maps/OpenDriveMap.OpenDriveMap:PersistentLevel.BP_Sky_2
LoadErrors: Warning: CreateExport: Failed to load Outer for resource 'MaterialInstanceDynamic_0': StaticMeshComponent /Game/Carla/Maps/OpenDriveMap.OpenDriveMap:PersistentLevel.BP_Sky_2.Planet
LoadErrors: Warning: CreateExport: Failed to load Outer for resource 'PostProcess': BP_Sky_C /Game/Carla/Maps/OpenDriveMap.OpenDriveMap:PersistentLevel.BP_Sky_2
LoadErrors: Warning: CreateExport: Failed to load Outer for resource 'SkyAtmosphere': BP_Sky_C /Game/Carla/Maps/OpenDriveMap.OpenDriveMap:PersistentLevel.BP_Sky_2
...
Assertion failed: ExternalEdgeOffset == ExternalEdges.Num() [File:./Developer/NaniteBuilder/Private/ClusterDAG.cpp] [Line: 126]

LogCore: Error: appError called: Assertion failed: ExternalEdgeOffset == ExternalEdges.Num() [File:./Developer/NaniteBuilder/Private/ClusterDAG.cpp] [Line: 126]

CommonUnixCrashHandler: Signal=11
Signal 11 caught
```

Blyron, 2025-08-21:
Have you tried with latest content?

anrp-tri, 2025-08-27:
I was specifically working with the 0.10.0 tag as I was rebasing our changes on to it (didn't want to target dev version); don't know if the tip of ue5-dev has the same issue.

Blyron, 2025-08-28:
In the 0.10.0 tag we cannot fix it. Ue5-dev doesn't have the same issue