carla-simulator/carla #8476: manual_control_with_traffic.py not starting

PabloVD, 2024-12-13:
The new script `manual_control_with_traffic.py` in `ue5-dev` which is a manual control with traffic generation in the same script does not work properly sometimes. Often, the pygame window opens but it freezes in black. The shell only outputs the first prints:
```
pygame 2.6.1 (SDL 2.28.4, Python 3.8.19)
Hello from the pygame community. https://www.pygame.org/contribute.html
INFO: listening to server 127.0.0.1:2000
Allows controlling a vehicle with a keyboard, generating traffic in the simulation
```

Blyron, 2024-12-17:
This is because the Minning Truck is trying to being spawned, the script must be run with --safe and filter a proper vehicle for Town 10

PabloVD, 2024-12-18:
Apparently that is not the reason, since the issue still happens after preventing the mining truck to spawn in Town 10. I checked that it sometimes gets stuck in the spawn pedestrians function, not sure why yet.

closed 2025-04-01 by Blyron