carla-simulator/carla #8481: Note for the release CARLA 0.10.0

Blyron, 2024-12-17:
Note for the release CARLA 0.10.0

We’ve conducted several profiling sessions on different PCs in the Package. As of today, the CPU seems to be the main bottleneck; the better the CPU, the better the results. The best we’ve achieved so far is 24/25 FPS (on Ubuntu 22) by adding the -RenderOffScreen flag to the server, essentially turning off the spectator camera.

[image]

As of now, we have no idea how to optimize further to reach 30 FPS. While it’s true that the other sensors can reach 30 FPS, the issue lies with the RGB Camera. The section highlighted in red shows the block where the camera sensor is being rendered, and the section in yellow corresponds to the spectator.

Yes, GPU parallelism is enabled (Image 2
[image]
), and you can see that the two command lists are being executed in parallel.
In Image 3, 
[image]
 there are several idle periods we don’t fully understand. In the first block, corresponding to the SceneCaptureComponent2D of the camera, you can see some gaps. We don’t know why they’re so large or what causes them. These gaps don’t occur in the second block, which also corresponds to the camera.

There are some long waits that we believe are due to the RenderThread waiting for the GPU (Image 4).

[image]

Here’s what we’ve tried so far:

Lowering the spectator quality without affecting the client (this hasn’t worked, and even if it did, the performance gain wouldn’t be sufficient as we’ve tested with spectator rendering completely disabled).
Using SceneCaptureComponent flags to speed up performance (this also didn’t work).

So we are open to suggestions and any contributions community can make.

lennyforterra, 2024-12-17:
General suggestions:

    Profiling the RenderThread:
        Use Unreal Engine’s GPU Visualizer and RenderDoc for deeper profiling.
        Identify the specific cause of the large idle periods.

    Experimenting with Camera Rendering:
        Test reducing SceneCapture resolution (if you haven't yet).
        Consider lowering post-processing quality specifically for RGB Camera.

    Thread Optimization:
        Investigate multi-threaded rendering settings in Unreal Engine to improve CPU-GPU synchronization, e.g.:
        -UseMultiThreading -numthreads=8 -r.RHICmdBypass=0 -s.AsyncLoadingThreadEnabled=1 -r.SceneRenderTargetResizeMethod=2

ZeKubiki, 2024-12-17:
General suggestions:

    Profiling the RenderThread:
        Use Unreal Engine’s GPU Visualizer and RenderDoc for deeper profiling.
        Identify the specific cause of the large idle periods.

    Experimenting with Camera Rendering:
        Test reducing SceneCapture resolution (if you haven't already).
        Consider lowering post-processing quality specifically for RGB Camera.

    Thread Optimization:
        Investigate multi-threaded rendering settings in Unreal Engine to improve CPU-GPU synchronization, e.g.:

Blyron, 2024-12-19:
We are aiming to get 30FPS with Scene capture resolution of 920x780. 
Thanks for the ideas! We will play with those

XLR8-VISION, 2025-01-14:
Disabling '**automatic tiling**' (if possible) will increase the speed of simulation. Automatic tiling is a part of multi-GPU feature.

1os3, 2025-07-30:
你们可以用5.6，感觉比5.3好，小版本升级不是很复杂

1os3, 2025-07-30:
> 你们可以用5.6，感觉比5.3好，小版本升级不是很复杂

当然比5.5也好

Blyron, 2025-07-30:
In english

Naxter, 2025-11-06:
Hey guys, is there any update to this? Have you found a solution to overcome this?