carla-simulator/carla #8485: Import customized map generated by Roadrunner using UE5 version

CAJCAJ, 2024-12-17:
Hi,

I already built up the latest UE5.5 version of CARLA on Windows 11. Now I am trying import the map files (.fbx and .xdor) generated by RoadRunner using the **make import**  but it doesn't work right now. Here is the snapshot for this issue:

[image]

Anyone can help import the map?

Thanks

Ji-Zhou, 2024-12-18:
Put your new map file and JSON configuration file in the import folder.

CAJCAJ, 2024-12-18:
Actually, I did try that and also used the '--no-carla-materials' tag without a JSON file, but neither worked

CAJCAJ, 2024-12-18:
> Put your new map file and JSON configuration file in the import folder.

Actually, I did try that and also used the '--no-carla-materials' tag without a JSON file, but neither worked

Ji-Zhou, 2024-12-19:
Check if your directory is correct.

Directly use "make import" to test.

you can share me more details about your directory and json file

Blyron, 2024-12-19:
Make import won't work in UE5 version

Blyron, 2025-01-02:
We haven't create a proper tool for UE5 importing content we will work on it

closed 2025-01-02 by Blyron

CAJCAJ, 2025-01-13:
> We haven't create a proper tool for UE5 importing content we will work on it

Thanks for your reply Blyron. I am also wondering if UE5 version support .Laz file import in the future?

ajay1606, 2025-12-19:
@Blyron Are there any update on this ? Looking for "import" custom map built in road runner to Carla with UE5.
Appreciate any help.