carla-simulator/carla #8486: Path planning failed —— Node 1 can not reachable from Node 2

Ji-Zhou, 2024-12-18:


Ji-Zhou, 2024-12-18:
Dear colleagues  

I have configured a custom map in Roadrunner. The roads in the custom map are connected by junctions, which have been configured.  
﻿
And some spawn points were set along the road in UE4 Carla.  
﻿
—Problem1: But when I plan my path in UE4 Carla, I often face failures—"Node 1 can not reachable from Node 2". It is interesting that sometimes node 1 can reach node 2 but node 2 cannot reach node 1.  
﻿
Specially, when i run the code—(route = grp.trace_route(origin, destination)). I will get successed result from SpawnPoint26 to 10, but failed from SpawnPoint 10 to 26.  
[image]
﻿
—Problem2: Another problem is that when I call traffic_generation.py, some autonomous vehicles seem unable to travel along the road and will cross the terrain between the two roads. As shown:  
[image]

﻿
—Problem3: I have set the starting and ending points that cannot be planned in problem 1 in the automatic_control.py. It seems to have found a route, but still faces the problem 2: intending to cross some infeasible terrain.  

Many thanks

leomilk-eng, 2025-01-16:
I know the reason for this problem. The reason why the planning cannot be done is that your road is not closed. The default planning is the A* algorithm, which causes an exception to be thrown when the road cannot be found. If you need a solution, please contact me.

leomilk-eng, 2025-01-16:
@Ji-Zhou I know the reason for this problem. The reason why the planning cannot be done is that your road is not closed. The default planning is the A* algorithm, which causes an exception to be thrown when the road cannot be found. If you need a solution, please contact me.

Ji-Zhou, 2025-01-16:
@leomilk-eng Thanks for your reply. I also discovered this reason. In fact, the road i need is not closed, and i need to do is to achieve path planning in such a scenario. 
When the starting and ending points are on lanes in different directions, the algorithm will fail because the road is not closed , and the default A * algorithm only searches along the specified direction of the road. I want to change this. Do you have any good ideas? I want to allow vehicles to cross roads in different directions while path planning!