carla-simulator/carla #8490: . /CarlaUE4.sh Illegal instruction (core dumped)

guilherme-verlab, 2024-12-19:
I tried to run CARLA 0.9.15 inside a container with Ubuntu 22.04, in a PC with Ubuntu 20.04, and I get this error:

`Illegal instruction (core dumped)`

How can I run CARLA in that way?

PabloVD, 2024-12-19:
If you want to run CARLA in a docker container, be sure that you has installed the [nvidia container toolkit](https://docs.nvidia.com/datacenter/cloud-native/container-toolkit/latest/install-guide.html) so that docker can access the gpu.

guilherme-verlab, 2024-12-19:
Hey @PabloVD, I already have nvidia container toolkit installed. I'm able to run CARLA 0.9.13 in a container, but the 0.9.15 version gives me that error

guilherme-verlab, 2024-12-19:
@PabloVD my nvidia container toolkit version: `NVIDIA Container Toolkit CLI version 1.16.2`

I ran the container with:
`docker run -it --runtime=nvidia -v /tmp/.X11-unix:/tmp/.X11-unix --gpus 'all,"capabilities=graphics,utility,display,video,compute"' --network=host --privileged -e DISPLAY=$DISPLAY --name carla_0_9_15 val_carla_0_9_15:latest bash`

Inside the container, I downloaded the precompiled package and extracted:
`wget https://tiny.carla.org/carla-0-9-15-linux && \`
 `mkdir -p /home/CARLA && \`
  `mv carla-0-9-15-linux /home/CARLA/ && \`
  `tar -xzf /home/CARLA/carla-0-9-15-linux -C /home/CARLA && \`
  `rm /home/CARLA/carla-0-9-15-linux`

I created a non-root user to run CARLA:
`yes | adduser --disabled-password --gecos "" carla_user && \`
`chown -R carla_user:carla_user /root && \`
`echo "carla_user ALL=(ALL) NOPASSWD:ALL" >> /etc/sudoers`

So, I followed the steps described in [https://carla.readthedocs.io/en/latest/start_quickstart/#b-package-installation](https://carla.readthedocs.io/en/latest/start_quickstart/#b-package-installation):
`apt update`
`apt install python3-pip`
`pip3 install --upgrade pip`
`pip3 install pygame numpy carla`

And tried to run Carla server with:
`cd /home/CARLA/`
`su carla_user`
`./CarlaUE4.sh`

And returned only this message:
`Illegal instruction (core dumped)`

This process works fine when I run CARLA 0.9.13.

asdacap, 2025-03-17:
In my case, I'm running carla in a VM (proxmox) and it did not forward a lot of cpu instruction. Checked `cat /proc/cpuinfo` to verify.