carla-simulator/carla #8495: Digital Twin Tool Fails to Import Maps from OpenStreetMap

Trans-cending, 2024-12-19:
Hi, everyone. I am interested in digital twin tool in Carla. I want to select a region from OpenStreetMap and paste the url to "UW_OnRoadMainWidget" and then click "Generate" button. **However, no matter which size of selected map I choose, the UE4 editor crashed with error. When I relaunch the UE4 editor and opened the generated XXX.umap and then run simulation, nothing (no buildings, vegetation and roads) is shown on the interface.** Can anyone help me? Thanks a lot. Problems are described in detail as follows.

CARLA version: 0.9.15 (build from source)
Platform/OS: Windows 10, UE 4.26
Steps to reproduce:

Note that osm_renderer is not available for Windows users. So I use the url to transform selected map according to [Digital Twin Tool](https://carla.readthedocs.io/en/latest/adv_digital_twin/).

1. First, I launch the "UW_OnRoadMainWidget" and paste the [OpenStreetMap](https://www.openstreetmap.org/export#map=17/1.356937/103.704274) url and click "Generate" button.
[image]
[image]

2. "UW_OnRoadMainWidget" starts generating maps for Carla. It seems normal at the beginning. But However, in the middle and later stages, the editor will become stick, and finally report an error and then **crash**.
[image]
[image]
   ("加载失败" means "fail to load"; "引用者" means "quoter" or "citer". The English version 
   of error is "fail to load /Game/Carla/Static/Pedestrian/Animations/GEN3/Nos_/AS_walkingG3.AS_walkingG3 quoter 
Default_BP_Walker_AB001_G3_C")

3. Then I look at the files generated under 'carla-0.9.15\Unreal\CarlaUE4\Content\CustomMaps' path. The structure is shown in the figure.
[image]
[image]

   
4. Lastly, I relaunch the UE4 editor and open the test.umap. I click the 'Run' button but only to find 
that no buildings, vegetation and roads are displayed, meaning that generation process fails.
[image]

5. I try to run manual_control.py to test the validity of customed map. And it raises 
```
INFO: listening to server 127.0.0.1:2000
ERROR: unable to parse the OpenDRIVE XML string
RuntimeError: failed to generate map
The server could not send the OpenDRIVE (.xodr) file:
Make sure it exists, has the same name of your town, and is correct.
```

If I run the default map (Town10), the whole process works normally. Thanks for your help.

EvilQQQ, 2024-12-22:
Same problem, but I successfully generate the map. However, it doesn't work when use PythonAPI. #8448

Trans-cending, 2024-12-23:
@EvilQQQ Hi, I have used LibCarla and CarlaUE4 in ue4-dev to replace the counterparts in carla-0.9.15. However, After many attempts, I find that if selected region has buildings, it will crash. Do you know how to fix it.

EvilQQQ, 2024-12-26:
Can you reproduce your error？I need more details such as Carsh log  Additionally, I have successfully generated the map and spawned vehicles on it.
[image]
[image]
Platfor: Windows10  0.9.15

Trans-cending, 2024-12-27:
@EvilQQQ Yeah, I can also generate maps like you. But you can notice that the buildings in the map is not that large. I try this region [url 1](https://overpass-api.de/api/map?bbox=118.788326,32.057168,118.790059,32.058714) but failed (UE4 editor crashed without error. There seems no error in log file). However, when I try this region [url 2](https://overpass-api.de/api/map?bbox=139.869202,35.716217,139.871562,35.717998) it works. My RAM is 32 GB and I also set the virtual memory.

llmmjj13, 2025-09-25:
i also meet this problem. all i did is similar to you， but there is nothing in the map. how to solve this problem？

[image]