carla-simulator/carla #8502: Failed to respond when control mining earth mover in Mine_01

donghaiwang, 2024-12-23:
CARLA version: 0.10.0
Platform/OS: Win10
Problem you have experienced: Failed to respond when running mining earth mover in Mine_01
What you expected to happen: control mining truck in map Mine_01
Steps to reproduce:
1) run `CarlaUnreal.exe`
2) switch map with [config.py](https://github.com/carla-simulator/carla/blob/ue5-dev/PythonAPI/util/config.py) 
```
python config.py -m Mine_01
```
3) run manual control with [manual_control.py](https://github.com/carla-simulator/carla/blob/ue5-dev/PythonAPI/examples/manual_control.py) 
```
python manual_control.py --filter vehicle.miningtruck.miningtruck
```
Failed to respond.
[image]

guilherme-verlab, 2024-12-24:
I had a similar issue, try to download the precompiled again and run. `--filter` should be `miningtruck`

VitaminQRQ, 2024-12-24:
Could you please let me know how you downloaded and extracted Carla v0.10.0? I got it from the Release page, but it says "The archive is corrupt" when I try to extract it. Could it be a download issue? 😢

donghaiwang, 2024-12-24:
I upload Carla v0.10.0 to [BaiduNet disk link](https://openhutb.github.io/carla_doc/) in `software/car/carla_release/0.10.0`.

[image]

donghaiwang, 2024-12-24:
> Could you please let me know how you downloaded and extracted Carla v0.10.0? I got it from the Release page, but it says "The archive is corrupt" when I try to extract it. Could it be a download issue? 😢

I upload Carla v0.10.0 to [BaiduNet disk link](https://openhutb.github.io/carla_doc/) in `software/car/carla_release/0.10.0/Carla-0.10.0-Win64-Shipping.zip`.

[image]

VitaminQRQ, 2024-12-24:
> I upload Carla v0.10.0 to [BaiduNet disk link](https://openhutb.github.io/carla_doc/) in `software/car/carla_release/0.10.0/Carla-0.10.0-Win64-Shipping.zip`.

Thanks a lot! 😁

geiwp, 2024-12-26:
I had a similar problem. Is the reason as follows:
WARNING: Client API version     = 0.9.15
WARNING: Simulator API version  = 0.10.0
INFO:  Could not find the required file in cache, downloading...  Carla/Maps/OpenDrive/Mine_01.xodr
WARNING: No InMemoryMap cache found. Setting up local map. This may take a while...
-------------------------------------------
Client API version =0.10.0,pip install carla==0.10.0 How to make the following error:
ERROR: Could not find a version that satisfies the requirement carla==0.10.0 (from versions: 0.9.5, 0.9.12, 0.9.13, 0.9.14, 0.9.15)
ERROR: No matching distribution found for carla==0.10.0

donghaiwang, 2024-12-26:
> ## I had a similar problem. Is the reason as follows:
> WARNING: Client API version = 0.9.15
> WARNING: Simulator API version = 0.10.0
> INFO: Could not find the required file in cache, downloading... [...]

We can install carla client with compied PythonAPI in Carla-0.10.0-Win64-Shipping.zip. (`Carla-0.10.0-Win64-Shipping\PythonAPI\carla\dist\carla-0.10.0-cp312-cp312-win_amd64.whl`)

PabloVD, 2025-01-10:
This issue was fixed and the packages updated, please download it again.
Be sure that you are using the carla python wheel from the 0.10 version, inside \PythonAPI\carla\dist
Install it via `pip install carla-0.10.0-cp312-cp312-win_amd64.whl`
Use the wheel corresponding to your python version.

closed 2025-01-10 by PabloVD