carla-simulator/carla #8503: Carla UE5 (precomplied) server not launching Ubuntu 22.04.5

sdhar16, 2024-12-23:
Thanks for contributing to CARLA!

If you are reporting an issue, please use the following outline:

CARLA version: 0.10.0
Platform/OS: Ubuntu 22.04.5
Problem you have experienced: ./CarlaUnreal.sh freezes and then crashes on launch
What you expected to happen: 
Steps to reproduce: Download the prebuild binary and start Carla
Other information (documentation you consulted, workarounds you tried):

Nvidia: 565.57.01
GPU: RTX 4090

I tried changing the ulimit -n 65535, but no effect

manne11, 2024-12-23:
Had a similar Issue, turned out that the gpu was running out of memory and crashing for me when lib-vulcan tried to pool it. You can monitor it through "nvidia-smi". Would be helpful to see what the error was.

MarcelPiNacy-CVC, 2025-01-02:
Hi,

could you share the crash log?

sdhar16, 2025-01-03:
There was no crash log, my pc kills the process and restarts the display manager. I think its because the VRAM gets filled. I find it interesting as I have a 24gb card.

MarcelPiNacy-CVC, 2025-01-03:
You should find the crash logs in the CarlaUnreal/Saved/Logs folder.

sdhar16, 2025-01-06:
Hi, 
I checked but the logs were not written during the crashes. I checked inside Carla and the logs in ./home/user/.config/Epic/CarlaUnreal/Saved/