carla-simulator/carla #8504: Carla UE5 (precomplied) ROS2 ISSUE

guilherme-verlab, 2024-12-24:
CARLA version: 0.10.0
Platform/OS: Ubuntu 22.04

The ros2_native example isn't working. When launching CARLA with:
`./CarlaUnreal.sh --ros2`

No ros2 nodes are initialized.

If run the `ros2_native.py`, topics send no data.

joel-mb, 2025-01-08:
Hey @guilherme-verlab 
When running the CARLA server with ROS2 enabled, no nodes are being initialized. However, upon starting the server, you should be able to see the `/clock` topic, even before running any scripts.

Could you provide more details to help me understand what might be happening? Are you seeing any warnings?

* Which is your development environment? Do you have ROS2 installed at system level or are you using docker?
* Is the vehicle correctly spawned when running the `ros2_native.py` script (i.e., can you see it in the spectator)?
* Are you seeing any topics when running `ros2 topic list`?
* Can you see the topics, but no data is being sent?

guilherme-verlab, 2025-01-08:
hey @joel-mb, now it's working, I've forgot to load the fastrtps-profile.xml file.

Markurt, 2025-02-05:
Hi, I am facing the same issue after loading the fastrtps. I am working on Ubuntu22.04 + ROS2 Humble. After run `./CarlaUnreal.sh --ros2`, not `/clock` topic shows. Any idea? @joel-mb

1157457050, 2025-02-06:
> Hi, I am facing the same issue after loading the fastrtps. I am working on Ubuntu22.04 + ROS2 Humble. After run `./CarlaUnreal.sh --ros2`, not `/clock` topic shows. Any idea? [@joel-mb](https://github.com/joel-mb)

I also encountered this problem, have you solved it?