carla-simulator/carla #8505: Running CarlaUnreal.exe crashed

chenmoc, 2024-12-24:
CARLA version: 0.10.0
Platform/OS: Windows 11 (22H2) 
Problem you have experienced: Running CarlaUnreal.exe crashed. The info is "The UE-CarlaUnreal game has crashed and will close."
What you expected to happen: Running CarlaUnreal.exe successful
Steps to reproduce: download and unzip Carla-0.10.0-Win64-Shipping.zip, then run CarlaUnreal.exe. 
Other information (documentation you consulted, workarounds you tried):
the log is below:
LowLevelFatalError [File:C:\UnrealEngine5_carla\Engine\Source\Runtime\RenderCore\Private\RenderingThread.cpp] [Line: 863] 
Rendering thread exception:
Fatal error!

Unhandled Exception: EXCEPTION_ACCESS_VIOLATION reading address 0x0000000000000018

0x00007ff63afb0b31 CarlaUnreal-Win64-Shipping.exe!ClearUnusedGraphResourcesImpl() [C:\UnrealEngine5_carla\Engine\Source\Runtime\RenderCore\Private\RenderGraphUtils.cpp:41]
0x00007ff63a8fc995 CarlaUnreal-Win64-Shipping.exe!Nanite::AddClearVisBufferPass() [C:\UnrealEngine5_carla\Engine\Source\Runtime\Renderer\Private\Nanite\NaniteCullRaster.cpp:5639]
0x00007ff63a91d3c2 CarlaUnreal-Win64-Shipping.exe!Nanite::InitRasterContext() [C:\UnrealEngine5_carla\Engine\Source\Runtime\Renderer\Private\Nanite\NaniteCullRaster.cpp:5773]
0x00007ff63abfc650 CarlaUnreal-Win64-Shipping.exe!FDeferredShadingSceneRenderer::AddSegmentationSensorPass() [C:\UnrealEngine5_carla\Engine\Source\Runtime\Renderer\Private\SegmentationSensorRendering.cpp:496]
0x00007ff63a5722e0 CarlaUnreal-Win64-Shipping.exe!FDeferredShadingSceneRenderer::Render() [C:\UnrealEngine5_carla\Engine\Source\Runtime\Renderer\Private\DeferredShadingRenderer.cpp:3355]
0x00007ff63ab98a87 CarlaUnreal-Win64-Shipping.exe!RenderViewFamilies_RenderThread() [C:\UnrealEngine5_carla\Engine\Source\Runtime\Renderer\Private\SceneRendering.cpp:5434]
0x00007ff63ab833db CarlaUnreal-Win64-Shipping.exe!UE::Core::Private::Function::TFunctionRefCaller<`FRendererModule::BeginRenderingViewFamilies'::`51'::<lambda_3>,void,FRHICommandListImmediate &>::Call() [C:\UnrealEngine5_carla\Engine\Source\Runtime\Core\Public\Templates\Function.h:321]
0x00007ff63afd1664 CarlaUnreal-Win64-Shipping.exe!`FRenderThreadCommandPipe::EnqueueAndLaunch'::`5'::<lambda_1>::operator()() [C:\UnrealEngine5_carla\Engine\Source\Runtime\RenderCore\Private\RenderingThread.cpp:1541]
0x00007ff639371500 CarlaUnreal-Win64-Shipping.exe!TGraphTask<TFunctionGraphTaskImpl<void __cdecl(void),0> >::ExecuteTask() [C:\UnrealEngine5_carla\Engine\Source\Runtime\Core\Public\Async\TaskGraphInterfaces.h:635]
0x00007ff639382541 CarlaUnreal-Win64-Shipping.exe!UE::Tasks::Private::FTaskBase::TryExecuteTask() [C:\UnrealEngine5_carla\Engine\Source\Runtime\Core\Public\Tasks\TaskPrivate.h:507]
0x00007ff63937df51 CarlaUnreal-Win64-Shipping.exe!FNamedTaskThread::ProcessTasksUntilQuit() [C:\UnrealEngine5_carla\Engine\Source\Runtime\Core\Private\Async\TaskGraph.cpp:667]
0x00007ff63afef242 CarlaUnreal-Win64-Shipping.exe!RenderingThreadMain() [C:\UnrealEngine5_carla\Engine\Source\Runtime\RenderCore\Private\RenderingThread.cpp:318]
0x00007ff63afef9f7 CarlaUnreal-Win64-Shipping.exe!FRenderingThread::Run() [C:\UnrealEngine5_carla\Engine\Source\Runtime\RenderCore\Private\RenderingThread.cpp:443]
0x00007ff6395396e7 CarlaUnreal-Win64-Shipping.exe!FRunnableThreadWin::Run() [C:\UnrealEngine5_carla\Engine\Source\Runtime\Core\Private\Windows\WindowsRunnableThread.cpp:159]
0x00007ff6395394b9 CarlaUnreal-Win64-Shipping.exe!FRunnableThreadWin::GuardedRun() [C:\UnrealEngine5_carla\Engine\Source\Runtime\Core\Private\Windows\WindowsRunnableThread.cpp:79]
0x00007ffae04e26ad KERNEL32.DLL!UnknownFunction []

chenmoc, 2024-12-24:
[image]

Bonjour321, 2025-01-02:
Same situation

MarcelPiNacy-CVC, 2025-01-02:
Hi,

could you share the output of the command `nvidia-smi`?

closed 2025-01-28 by MarcelPiNacy-CVC

Abhinav-2901, 2025-02-07:
> Hi, I'm closing this issue for now. Feel free to reopen it if this persists.

C:\Users\arush>nvidia-smi
Fri Feb  7 19:55:53 2025
+-----------------------------------------------------------------------------------------+
| NVIDIA-SMI 561.19                 Driver Version: 561.19         CUDA Version: 12.6     |
|-----------------------------------------+------------------------+----------------------+
| GPU  Name                  Driver-Model | Bus-Id          Disp.A | Volatile Uncorr. ECC |
| Fan  Temp   Perf          Pwr:Usage/Cap |           Memory-Usage | GPU-Util  Compute M. |
|                                         |                        |               MIG M. |
|=========================================+========================+======================|
|   0  NVIDIA GeForce RTX 3050 ...  WDDM  |   00000000:01:00.0 Off |                  N/A |
| N/A   31C    P0              9W /   60W |       0MiB /   4096MiB |      0%      Default |
|                                         |                        |                  N/A |
+-----------------------------------------+------------------------+----------------------+

+-----------------------------------------------------------------------------------------+
| Processes:                                                                              |
|  GPU   GI   CI        PID   Type   Process name                              GPU Memory |
|        ID   ID                                                               Usage      |
|=========================================================================================|
|  No running processes found                                                             |
+-----------------------------------------------------------------------------------------+

I am facing same issue.
need help urgently