carla-simulator/carla #8506: Ubuntu Building UE5 Error

Errorsteved, 2024-12-24:
CARLA version: latest cloned Carla ue5-dev version
Platform/OS: Ubuntu24.04
Problem you have experienced: fail building and split a lot of errors like :
`[272/2308] Compile Module.Learning.cpp
In file included from /home/steve/Xeon/dev/carla/UnrealEngine5_carla/Engine/Plugins/Experimental/LearningAgents/Intermediate/Build/Linux/x64/UnrealEditor/Development/Learning/Module.Learning.cpp:7:
/home/steve/Xeon/dev/carla/UnrealEngine5_carla/Engine/Plugins/Experimental/LearningAgents/Source/Learning/Private/LearningAdamOptimizer.cpp:8:10: fatal error: 'Learning.ispc.generated.h' file not found
    8 | #include "Learning.ispc.generated.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~ `

What you expected to happen: Build UE successfully
Steps to reproduce: Use Ubuntu, Clone latest dev-ue5, Run CarlaSetup.sh. And will occur when compiling Unreal files.
Other information (documentation you consulted, workarounds you tried): I’ve followed “[UE 5.5 Source Linux Compilation Errors - #2 by NuLL3rr0r](https://forums.unrealengine.com/t/ue-5-5-source-linux-compilation-errors/2148769/2)” and added the following lines to my Game and Editor’s .Target.cs and rebuild:
```

// Disable Intel Implicit SPMD Program Compiler forced by UE 5.5 on Linux.
    bCompileISPC = false;
```
However, these errors occur randomly. For instance, one time it might be “fatal error: ‘Learning.ispc.generated.h’ file not found,” and the next time it could be “fatal error: ‘NNERuntimeBasicCpu.ispc.generated.h’ file not found.”

jhallier, 2025-01-03:
I have the same issues in Ubuntu 24.04. Has anyone built successfully under Ubuntu 24.04 yet? I already switched to the UE5 version as the UE4 explicitly excluded Ubuntu 24.04, and I did not really want to downgrade or install an outdated version of Ubuntu only for Carla (even 20.04 is losing its long-term support in 3 months from now, and 22.04 is only somewhat "experimental" for the UE4 version.

Btw it seems that the "randomness" may simply be in the order of compilation of files? It seems that all ispc files are affected. Strange that it still does not work with ispc disabled (I haven't tried it), and why it would be a Ubuntu 24.04 issue?

manne11, 2025-04-16:
Got the same issues on Ubuntu 24.04 . What is bizarre is that it complains about:
`[157/4244] Compile Module.SignalProcessing.1.cpp
In file included from /home/manne/projects/UnrealEngine5_carla/Engine/Intermediate/Build/Linux/x64/UnrealEditor/Development/SignalProcessing/Module.SignalProcessing.1.cpp:40:
/home/manne/projects/UnrealEngine5_carla/Engine/Source/Runtime/SignalProcessing/Private/FloatArrayMath.cpp:15:10: fatal error: 'FloatArrayMath.ispc.generated.h' file not found
   15 | #include "FloatArrayMath.ispc.generated.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
`

but I see that it is generated a few steps just before that 

`[142/4244] Generate Header [x86-64] FloatArrayMath.ispc
[143/4244] Copy [x86-64] FloatArrayMath.ispc.generated.h
`

Could you confirm that you see a similar behavior? Then it may be an issue of timing or mismatched location.

manne11, 2025-04-16:
Was able to solve it. I ran "make UnrealEditor"(since the ispc errors were from compiling the unrealeditor) after the failed compilation and it compiled the modules that failed. It turned to be a timing issue,  the files were not available when it needed it. I also installed `ispc(1.22.0)`, I'm not sure if that helped compile them though.  

Also checked by running the unrealeditor binary.

Let me know if that worked for you!

CarlaEnjoyer, 2025-04-21:
It worked, thanks! ispc(1.22.0) was not needed from my part.

qhaas, 2025-05-03:
Seeing the same issue in Ubuntu 24.04 and [Carla's UnrealEngine fork commit eea6a8e677e3509a3e56122d69e3d380bb624bbf](https://github.com/CarlaUnreal/UnrealEngine/commit/eea6a8e677e3509a3e56122d69e3d380bb624bbf).

Error:
```
[1617/4244] Compile Module.Learning.cpp
In file included from /mnt/unreal-engine-src/Engine/Plugins/Experimental/LearningAgents/Intermediate/Build/Linux/x64/UnrealEditor/Development/Learning/Module.Learning.cpp:7:
/mnt/unreal-engine-src/Engine/Plugins/Experimental/LearningAgents/Source/Learning/Private/LearningAdamOptimizer.cpp:8:10: fatal error: 'Learning.ispc.generated.h' file not found
    8 | #include "Learning.ispc.generated.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~
1 error generated.
```

Dockerfile:
```
FROM [${BASE_IMAGE} AS base](ubuntu:24.04)

# use UTC as default timezone
ENV TZ=Etc/UTC
# debian-like images default to ascii
ENV LANG=C.UTF-8

# from https://github.com/carla-simulator/carla/blob/ue5-dev/Util/SetupUtils/InstallPrerequisites.sh
ARG APT_PACKAGES="\
    build-essential \
    make \
    ninja-build \
    libvulkan1 \
    libpng-dev \
    libtiff5-dev \
    libjpeg-dev \
    tzdata \
    sed \
    curl \
    libtool \
    rsync \
    libxml2-dev \
    git \
    git-lfs \
    python3 \
    python3-dev \
    python3-pip \
    cmake \
"
ARG DEBIAN_FRONTEND=noninteractive
RUN apt-get update && apt-get install -y $APT_PACKAGES && apt-get clean
```

Commands:
```
./Setup.sh
./GenerateProjectFiles.sh
make
```

manne11, 2025-05-04:
Just an addendum. Something I realised after reading the docs is that Ubuntu 24 is not supported. It ran flawlessly on Ubuntu 20, none of these errors were raised.

qhaas, 2025-05-04:
> Ubuntu 24 is not supported. It ran flawlessly on Ubuntu 20, none of these errors were raised.

[Ubuntu 20.04 is EOL this month](https://ubuntu.com/blog/ubuntu-20-04-eol-for-devicesional), so we likely need to move on.  I notice that the UE5 build toolchain is now based on EL8 (specifically, Rocky Linux 8).  I rolled back to 22.04 inside a container and built CARLA"s UE5 fork without having to run `make` twice.