carla-simulator/carla #8507: Windows11  CARLA0.10.0

geiwp, 2024-12-24:
I had the following problems after downloading and unziping CARLA0.10.0:
[image]
导致无法打开CarlaUnreal.exe.
[image]
How to solve this problem?

VitaminQRQ, 2024-12-24:
Same problem. BandZip reported "The archive is corrupt." 😢

geiwp, 2024-12-24:
[image]

IttaiGitGet, 2024-12-24:
Unzipping fails.
On Win11 version 23H2 OS build 22631.4602. When unzipping Carla-0.10.0-Win64-Shipping.zip, 7Zip reports: 
1. "Unexpected end of data" .
2. "There are some data after the end of the payload data: Carla-0.10.0-Win64-Shipping\CarlaUnreal\Binaries\Win64\CarlaUnreal-Win64-Shipping.exe

VitaminQRQ, 2024-12-24:
This link can help: https://github.com/carla-simulator/carla/issues/8502#issuecomment-2561132628

Basavaraj-PN, 2025-01-07:
Same problem

Basavaraj-PN, 2025-01-07:
Same problem

WuJieLuen, 2025-01-15:
same problem

WuJieLuen, 2025-01-15:
I have solve this isuue by using VPN, I jump to US site and download from the official link by Microsoft Edge browser.
It's work! but I don't know why.

closed 2025-04-01 by Blyron