carla-simulator/carla #8510: The output of RGB camera and instance segmentation camera in carla0.9.14 is different

DarkerSun6, 2024-12-27:
Hello, I have placed an RGB camera and an instance segmentation camera in the same location on the TOWN 01 map in version 0.9.14 of Carla.
There is a problem here, the mountains that can be seen in the RGB camera are missing in the instance segmentation camera.
And the vehicles hidden behind the mountain cannot be seen in the RGB camera, but are seen in the instance segmentation camera.
May I ask what the problem is
[image]
[image]

lynrookie, 2024-12-30:
The semantic segmentation camera renders elements in scene with a different color depending on how these have been tagged. The tags are created by the simulator depending on the path of the asset used for spawning. For example, meshes tagged as Pedestrians are spawned with content stored in "Unreal/CarlaUE4/Content/Static/Pedestrians".
So, back to your problem, the mountains that can be seen in the RGB camera are missing in the instance segmentation camera, 
And the vehicles hidden behind the mountain can not be seen in the RGB camera, but are seen in the instance segmentation camera. 
That is because the mountains asset is not tagged in right way, You may check it in the [CARLA documentation](https://carla.readthedocs.io/en/latest/tuto_G_retrieve_data/#semantic-segmentation-camera)

Blyron, 2025-01-02:
Right now, the CARLA UE4 version does not support landscapes

tdegraaff, 2025-01-30:
@jorge-kabuto Can now be closed (https://github.com/carla-simulator/carla/pull/8528).

closed 2025-01-31 by Blyron