carla-simulator/carla #8511: Vehicle not sensing stones and debris in Mine_01 map

darshanAtSwaayatt, 2024-12-27:
CARLA version: 0.10.0
Platform/OS: Ubuntu 22.04.5
Problem you have experienced: While using the Mine_01 map, the vehicle does not sense the debris or stones in the road, it passes through it
What you expected to happen: As observed in Town 10, the vehicle senses the speed breakers and debris and behaves accordingly. Same should happen in the Mine_01 map.
Steps to reproduce: Move a vehicle in the Mine_01 map, change the camera to a side view so you can see the vehicle tyres, and move the vehicle on the stones/debris.
Other information (documentation you consulted, workarounds you tried): I could not find any documentation to enable such behavior

Blyron, 2025-01-02:
We will share that with the map provider

yuanhaorannnnnn, 2025-01-10:
> We will share that with the map provider

We will update mine map to fix related issues next week.

yuanhaorannnnnn, 2025-01-16:
We're gonna update the map to fix this issue.

darshanAtSwaayatt, 2025-01-27:
> We're gonna update the map to fix this issue.

Has the mine map been updated?

darshanAtSwaayatt, 2025-02-24:
> > We're gonna update the map to fix this issue.
> 
> Has the mine map been updated?

Updates please.

Blyron, 2025-02-24:
Instance Segmentation has been updated properly. About rocks sensing still on going. You can try by adjusting vehicles wheels trace by editing sweep_type

Blyron, 2025-03-28:
This should be fixed in latest ue5-dev content commit. Thanks for your patience

closed 2025-03-28 by Blyron