carla-simulator/carla #8513: Can't load other carla towns in carla 0.10 UE5 version

abdohamdy7, 2024-12-29:
8 world = client.load_world('Town01')
      9 print("Connected to Carla server and loaded Town 1")

RuntimeError: std::exception

Steps to reproduce:
import carla

# Connect to the Carla simulator server
client = carla.Client('localhost', 2000)
client.set_timeout(10.0)

# Load Town 1
world = client.load_world('Town01')

-->

GoodarzMehr, 2024-12-30:
I think 0.10.0 only supports Town10HD for the time being, as explained in the release notes.

abdohamdy7, 2024-12-30:
Is there a way to add any other town? They say that the other towns are not upgraded. But I don't want them upgraded.
In addition, in the new documentation,  they provided a way to change the map and provided an example of town01.

GoodarzMehr, 2024-12-30:
I'm not sure if the old maps would work without upgrading because my understanding is that UE5 has a completely different approach to lighting and rendering. Someone from the CARLA team can perhaps elaborate on this.

That part in the documentation is just an example of how to change maps.

abdohamdy7, 2024-12-30:
[like]  abdo hamdy reacted to your message:

Minia University https://www.minia.edu.eg/ ---[MU Ranking :Times: 600+ / QS : 81-90 (Arab Region) US Ranking: 1259 / Shanghhai: 401-500 Pharmacy & Pharmaceutical Sciences]

abdohamdy7, 2024-12-30:
[0]     abdo hamdy reacted to your message:

Blyron, 2025-01-02:
You can do it in editor mode. In Binaries you cannot, so you will need to compile the editor and do the whole setup