carla-simulator/carla #8516: Encountered an error while installing the carla library in a virtual environment

wensui123, 2024-12-31:
I created a virtual environment in Anaconda and installed Python 3.12 in the environment. Then, I used the ‘pip install carla’ command to install the carla library, but I encountered the following error：

Collecting carla
  Using cached carla-0.9.5.tar.gz (3.9 kB)
  Preparing metadata (setup.py) ... error
  error: subprocess-exited-with-error

  × python setup.py egg_info did not run successfully.
  │ exit code: 1
  ╰─> [10 lines of output]
      Traceback (most recent call last):
        File "<string>", line 2, in <module>
        File "<pip-setuptools-caller>", line 34, in <module>
        File "C:\Users\*****\pip-install-otvmw0xp\carla_37c4c7f3e9b34f0bbae37e8addf13acc\setup.py", line 114, in <module>
          ext_modules=get_libcarla_extensions(),
                      ^^^^^^^^^^^^^^^^^^^^^^^^^
        File "C:\Users\*****\pip-install-otvmw0xp\carla_37c4c7f3e9b34f0bbae37e8addf13acc\setup.py", line 75, in get_libcarla_extensions
          libs = [x for x in os.listdir('dependencies/lib') if any(d in x for d in required_libs)]
                             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
      FileNotFoundError: [WinError 3] 系统找不到指定的路径。: 'dependencies/lib'
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.

The Carla version I am using is 0.10.0. I do not know how to solve the problem.

TikaToka, 2025-01-03:
It might not related to problem, but pip is trying to use 0.9.15 cached to install, but you mentioned that you are using 0.10.0.

MarcelPiNacy-CVC, 2025-01-03:
> It might not related to problem, but pip is trying to use 0.9.15 cached to install, but you mentioned that you are using 0.10.0.

Sounds like this may be the source of the problem.

joel-mb, 2025-01-08:
@wensui123 The wheel files for CARLA 0.10.0 have not been released.

In your use case, when attempting to install `carla` via `pip`, as there is not public wheel that satisfies your requirements (Python 3.12), it selects the latest available source (in this case, 0.9.5) and tries to build it from source.

For CARLA 0.10.0, you can find the wheel files in the `PythonAPI/carla/dist` folder of the package. You can install it with `pip install <WHEEL_FILE>`

wensui123, 2025-01-08:
> @wensui123 The wheel files for CARLA 0.10.0 have not been released.
>
> In your use case, when attempting to install `carla` via `pip`, as there is not public wheel that satisfies your requirements [...]

Thank you,

wensui123, 2025-01-08:
@joel-mb ,yes, you are right. I used the carla wheel file in this folder.The problem had solved.Thank you very much😄。I will close the invitation.

closed 2025-01-08 by wensui123