carla-simulator/carla #8518: Compile UE5.5 on Ubuntu 22.04 reports an error. ld.lld: error: undefined symbol: UUserToolBoxBaseBlueprint::Execute

1157457050, 2025-01-02:


1157457050, 2025-01-02:
ld.lld: error: undefined symbol: UUserToolBoxBaseBlueprint::Execute()
>>> referenced by Module.UserToolBoxCore.cpp
>>>               ../Plugins/Experimental/UserToolBoxCore/Intermediate/Build/Linux/x64/UnrealEditor/Development/UserToolBoxCore/Module.UserToolBoxCore.cpp.o:(vtable for UUserToolBoxBaseBlueprint)
clang++: error: linker command failed with exit code 1 (use -v to see invocation)

MarcelPiNacy-CVC, 2025-01-02:
Hi,

Please provide the full build log as text.

closed 2025-01-08 by 1157457050

uuauto2024, 2025-01-08:
> ld.lld: error: undefined symbol: UUserToolBoxBaseBlueprint::Execute()

hi, have you solve this problem? I am facing same issue now

hbertin1, 2025-02-06:
Hello, I am facing the same issue while building UE5.5.2 on fedora. Does someone have a fix?

JohnGitinGud, 2025-12-30:
I got the exact same issue + more on Arch Linux 

> [1/6] Link (lld) libUnrealEditor-UserToolBoxCore.so
ld.lld: error: undefined symbol: UUserToolBoxBaseBlueprint::Execute()
>>> referenced by Module.UserToolBoxCore.cpp
>>>               /home/jc/Games/Unreal-Engine/UnrealEngine-fork/Engine/Plugins/Experimental/UserToolBoxCore/Intermediate/Build/Linux/x64/UnrealEditor/Development/UserToolBoxCore/Module.UserToolBoxCore.cpp.o:(vtable for UUserToolBoxBaseBlueprint)
clang++: error: linker command failed with exit code 1 (use -v to see invocation)
[2/6] Compile SharedMemoryMediaInitializerFeature.cpp
/home/jc/Games/Unreal-Engine/UnrealEngine-fork/Engine/Plugins/Runtime/nDisplay/Source/SharedMemoryMediaEditor/Private/ModularFeatures/SharedMemoryMediaInitializerFeature.cpp:3:9: error: #pragma once in main file [-Werror,-Wpragma-once-outside-header]
#pragma once

If we can somehow pass a -v flag in there, maybe the invocation have some clue?