carla-simulator/carla #8521: Carla UE5 (precomplied) ROS2 NOT WORKING

guilherme-verlab, 2025-01-02:
CARLA version: 0.10.0
Platform/OS: Ubuntu 22.04

The ros2_native example isn't working. When launching CARLA with:
`./CarlaUnreal.sh --ros2`

No ros2 nodes are initialized.

If run the `ros2_native.py`, topics send no data.

In the extracted CARLA folder, I didn't see any ros2 package.

closed 2025-01-07 by guilherme-verlab

awiswasi, 2025-03-21:
I am facing the same issue. Were you able to resolve it?
I see no instructions or examples about ROS2 anywhere, which is frustrating.

guilherme-verlab, 2025-03-21:
> I am facing the same issue. Were you able to resolve it? I see no instructions or examples about ROS2 anywhere, which is frustrating.

I've forgot to load the fastrtps-profile.xml file in PythonAPI/examples/ros2/config. Now it's working

awiswasi, 2025-03-22:
How do you load it?
I got the RViz to work but no ROS topics being published.

guilherme-verlab, 2025-03-22:
> How do you load it? I got the RViz to work but no ROS topics being published.

by 
`export FASTRTPS_DEFAULT_PROFILES_FILE=/path/to/PythonAPI/examples/ros2/config/fastrtps-profile.xml`

awiswasi, 2025-03-31:
Here is my process:
Terminal 1:
1. `source /opt/ros/humble/setup.bash`
2. `export FASTRTPS_DEFAULT_PROFILES_FILE=/opt/carla-ue5-dev/Build/Package/Carla-0.10.0-Linux-Shipping/PythonAPI/examples/ros2/config/fastrtps-profile.xml`
2. `./CarlaUnreal.sh --ros2`

Terminal 2:
1. `source /opt/ros/humble/setup.bash`
2. `export FASTRTPS_DEFAULT_PROFILES_FILE=/opt/carla-ue5-dev/Build/Package/Carla-0.10.0-Linux-Shipping/PythonAPI/examples/ros2/config/fastrtps-profile.xml`
3. `python3 ros2_native.py -f stack.json`

I get this error:
`INFO: Listening to server localhost:2000
Traceback (most recent call last):
  File "/opt/carla-ue5-dev/Build/Package/Carla-0.10.0-Linux-Shipping/PythonAPI/examples/ros2/ros2_native.py", line 131, in <module>
    main(args)
  File "/opt/carla-ue5-dev/Build/Package/Carla-0.10.0-Linux-Shipping/PythonAPI/examples/ros2/ros2_native.py", line 91, in main
    vehicle = _setup_vehicle(world, config)
  File "/opt/carla-ue5-dev/Build/Package/Carla-0.10.0-Linux-Shipping/PythonAPI/examples/ros2/ros2_native.py", line 29, in _setup_vehicle
    return  world.spawn_actor(
RuntimeError: std::exception`

Am I missing something?