carla-simulator/carla #8522: UE5 Editor's UI doesn't respond after starting the engine with the "cmake --build Build --target launch" command

utkuoguzman, 2025-01-03:
CARLA version:                          0.10.0
Platform/OS:                             Ubuntu 22.04.5
Problem you have experienced: After opening it with the target command (the second time) the UI doesn't respond.
What you expected to happen:  I want the UI to respond.
Steps to reproduce:                   Build carla 0.10.0 in Ubuntu 22.04.5. Let the UE5 open. Then close, then start it with "cmake --build Build --target launch". Realize that UI doesn't respond.
Other information (documentation you consulted, workarounds you tried): None. UI never came. But I also tried to install the unreal 5.5.1 without any Carla on my Ubuntu 22.04.5 and it had a similar problem as well. I don't know why. Please help.

Blyron-AeroSim, 2025-01-03:
Could you share the log after the UI is open?

Blyron, 2025-01-03:
Could you share the log after UI is open?

utkuoguzman, 2025-01-08:
I believe this is the relevant rows:
[2025.01.08-12.34.56:005][607]LogToolMenus: Warning: Menu item not found: 'OpenMarketplace' for insert: 'ContentBrowser'
[2025.01.08-12.35.12:103][265]LogToolMenus: Warning: Menu item not found: 'OpenMarketplace' for insert: 'ContentBrowser'
Do you want me to paste all logs?

utkuoguzman, 2025-01-13:
[2025.01.13-06.32.51:044][ 89]LogEOSSDK: Warning: LogEOS: Error response received from backend. Response is not JSON. ServiceName=[SDKConfig], OperationName=[GetPlatformConfigRoute], Url=[<Redacted>], HttpStatus=[403], CorrId=[]
[2025.01.13-06.32.51:064][ 90]LogEOSSDK: Warning: LogEOS: SDK Config Platform Update Request Failed, Result Code: EOS_UnrecognizedResponse, Retrying after 12.149671 seconds

utkuoguzman, 2025-01-13:
[2025.01.13-06.22.57:092][  3]LogPSOHitching: Encountered 50 PSO creation hitches so far (9 graphics, 41 compute). 0 of them were precached.
[2025.01.13-06.22.57:459][  4]LogSourceControl: Uncontrolled asset enumeration finished in 1.348739 seconds (Found 52282 uncontrolled assets)
[2025.01.13-06.22.57:815][  7]LogPSOHitching: Encountered 100 PSO creation hitches so far (18 graphics, 82 compute). 0 of them were precached.
[2025.01.13-06.22.58:522][ 11]LogPSOHitching: Encountered 150 PSO creation hitches so far (29 graphics, 121 compute). 0 of them were precached.
[2025.01.13-06.23.01:230][ 19]LogEOSSDK: LogEOS: Updating Platform SDK Config, Time: 21.582390
[2025.01.13-06.23.03:205][ 21]LogEOSSDK: Warning: LogEOS: Error response received from backend. Response is not JSON. ServiceName=[SDKConfig], OperationName=[GetPlatformConfigRoute], Url=[<Redacted>], HttpStatus=[403], CorrId=[]
[2025.01.13-06.23.04:192][ 22]LogEOSSDK: Warning: LogEOS: SDK Config Platform Update Request Failed, Result Code: EOS_UnrecognizedResponse, Retrying after 6.188087 seconds
[2025.01.13-06.23.06:960][ 25]LogEOSSDK: Error: LogEOS: EOS_Auth_CopyUserAuthToken - One or more parameters are null
[2025.01.13-06.23.06:960][ 25]LogFab: Error: User auth token is invalid - unable to get auth token
[2025.01.13-06.23.09:916][ 28]LogEOSSDK: LogEOS: Updating Platform SDK Config, Time: 30.268555

utkuoguzman, 2025-01-13:
[2025.01.13-06.22.42:765][  0]LogStreaming: Warning: Failed to read file 'Common/Selector.png' error.

utkuoguzman, 2025-01-13:
[2025.01.13-06.22.39:595][  0]LogStreaming: Warning: Failed to read file '../../../Engine/Plugins/Developer/PlasticSourceControl/Resources/Icon128.png' error.
[2025.01.13-06.22.39:595][  0]LogSlate: Could not find file for Slate resource: ../../../Engine/Plugins/Developer/PlasticSourceControl/Resources/Icon128.png
[2025.01.13-06.22.39:609][  0]LogStreaming: Warning: Failed to read file '../../../Engine/Plugins/Interchange/Editor/Content/Old/Tiles/Outer/alertSolid.png' error.
[2025.01.13-06.22.39:609][  0]LogSlate: Could not find file for Slate resource: ../../../Engine/Plugins/Interchange/Editor/Content/Old/Tiles/Outer/alertSolid.png
[2025.01.13-06.22.39:640][  0]LogEOSSDK: LogEOS: [Boot] EOSSDK Version 1.16.4-36651368 booting at 2025-01-13T06:22:39.640Z using C
[2025.01.13-06.22.39:640][  0]LogEOSSDK: LogEOS: [Boot] EOSSDK Platform Properties [OS=Linux/ubuntu22.04, ClientId=xyza7891REBVsEqSJRRNXmlS7EQHM459, ProductId=86f32f1151354e7cb39c12f8ab2c22a3, SandboxId=a01927f7421a4d4995673fe30ef46945, DeploymentId=a652a72ea1664dcab3a467891eea5f30, ProductName=CarlaUnreal, ProductVersion=UE5-CL-0, IsServer=false, Flags=DisableOverlay]
[2025.01.13-06.22.39:640][  0]LogEOSSDK: LogEOSAnalytics: Start Session (User: ...)
[2025.01.13-06.22.39:647][  0]LogEOSSDK: LogEOSAntiCheat: [AntiCheatClient] Anti-cheat client not available. Verify that the game was started using the anti-cheat bootstrapper if you intend to use it.
[2025.01.13-06.22.39:648][  0]LogEOSSDK: LogEOS: SetApplicationStatus - OldStatus: EOS_AS_Foreground, NewStatus: EOS_AS_Foreground, Current Time: 0001.01.01-00.00.00
[2025.01.13-06.22.39:648][  0]LogEOSSDK: LogEOS: SetNetworkStatus - OldStatus: EOS_NS_Online, NewStatus: EOS_NS_Online
[2025.01.13-06.22.39:648][  0]LogEOSSDK: LogEOS: Updating Platform SDK Config, Time: 0.000097
[2025.01.13-06.22.39:648][  0]LogFab: Display: Logging in using persist
[2025.01.13-06.22.39:718][  0]LogUObjectArray: 59038 objects as part of root set at end of initial load.
[2025.01.13-06.22.39:718][  0]LogUObjectArray: CloseDisregardForGC: 0/0 objects in disregard for GC pool
[2025.01.13-06.22.39:734][  0]LogStreaming: Display: AsyncLoading2 - NotifyRegistrationComplete: Registered 37572 public script object entries (1014.16 KB)
[2025.01.13-06.22.39:734][  0]LogStreaming: Display: AsyncLoading2 - Thread Started: false, IsInitialLoad: false
[2025.01.13-06.22.39:838][  0]LogClass: Error: BoolProperty FFabAssetMetadata::IsQuixel is not initialized properly. Module:Fab File:Private/FabBrowserApi.h
[2025.01.13-06.22.39:838][  0]LogClass: Display: 1 Uninitialized script struct members found including 0 object properties
[2025.01.13-06.22.39:890][  0]LogAutomationTest: Error: LogClass: BoolProperty FFabAssetMetadata::IsQuixel is not initialized properly. Module:Fab File:Private/FabBrowserApi.h

utkuoguzman, 2025-01-13:
[2025.01.13-06.22.36:679][  0]LogCarla: == CARLA Settings ==============================================================
[2025.01.13-06.22.36:679][  0]LogCarla: Last settings file loaded: 
[2025.01.13-06.22.36:679][  0]LogCarla: [CARLA/Server]
[2025.01.13-06.22.36:679][  0]LogCarla: RPC Port = 2000
[2025.01.13-06.22.36:679][  0]LogCarla: Streaming Port = 2001
[2025.01.13-06.22.36:679][  0]LogCarla: Secondary Port = 2002
[2025.01.13-06.22.36:679][  0]LogCarla: Synchronous Mode = Disabled
[2025.01.13-06.22.36:679][  0]LogCarla: Rendering = Enabled
[2025.01.13-06.22.36:679][  0]LogCarla: [CARLA/QualitySettings]
[2025.01.13-06.22.36:679][  0]LogCarla: Quality Level = Epic
[2025.01.13-06.22.36:679][  0]LogCarla: ================================================================================
[2025.01.13-06.22.36:766][  0]LogScript: Warning: Script Msg: A null object was passed as a world context object to UEngine::GetWorldFromContextObject().
[2025.01.13-06.22.36:767][  0]LogScript: Warning: Script Msg: A null object was passed as a world context object to UEngine::GetWorldFromContextObject().
[2025.01.13-06.22.36:773][  0]LogScript: Warning: Script Msg: A null object was passed as a world context object to UEngine::GetWorldFromContextObject().
[2025.01.13-06.22.36:793][  0]LogScript: Warning: Script Msg: A null object was passed as a world context object to UEngine::GetWorldFromContextObject().

utkuoguzman, 2025-01-13:
[2025.01.13-06.22.31:015][  0]LogRHI: Warning: Dynamic Uniform Buffers are enabled, but they will not be used with Vulkan bindless.
[2025.01.13-06.22.31:015][  0]LogRHI: Warning: Memory defrag is enabled, but it will not be used with Vulkan bindless.

Blyron, 2025-01-13:
Try to update libvulkan and this may be related with your harddisk being slow. We cannot support you as this issue comes from Unreal, try in Epic Games forum. I am closing this issue as it is not CARLA related

closed 2025-01-13 by Blyron