carla-simulator/carla #8526: Requirements for Carla in PC

manasvinisrini, 2025-01-04:
Hello,
I've been trying to buy the right PC to run CARLA on and I want to run simulations that aren't lagging or slow. 
Would these specs fit the requirement ? I'm a newbie at this and any advice would really help or alternatives. 
I'm not sure if the processor would be good enough :

 Intel Core i7 4th Gen. (up to 4.0 GHz)
NVIDIA GeForce GTX 1070
RAM - 32 GB
Storage - 1 TB SSD + 2 TB HDD
Maximum RAM - 64 GB

Would really appreciate any advice. Thanks a lot !

GoodarzMehr, 2025-01-05:
Depends on the types of simulation you want to run. Generally speaking, the more sensors and actors there are in a scene and the bigger the map is the slower the simulation gets, so better GPU and single-threaded CPU performance improve the simulation.

I would also recommend getting something with more VRAM than a 1070 if you wanna do more than the bare minimum.

MarcelPiNacy-CVC, 2025-01-07:
Is this UE4 or UE5?

tensorturtle, 2025-01-11:
For me, Carla 0.10.0 (UE5) fails to launch. Carla 0.9.15 ran without issues.
I have Intel Core i7 (12th gen), NVIDIA GeForce RTX2070 Super (8GB VRAM), and 32GB RAM.

It seems to be a Vulkan memory error? This is the crash output of `/.CarlaUnreal.sh`:
```
Shutdown handler: initialize.
5.5.0-0+UE5 1013 0
Disabling core dumps.
LowLevelFatalError [File:./Runtime/VulkanRHI/Private/VulkanMemory.cpp] [Line: 1995]
Out of memory on Vulkan; MemoryTypeIndex=1, AllocSize=128.000MB
Signal 11 caught.
Malloc Size=262146 LargeMemoryPoolOffset=262162
CommonUnixCrashHandler: Signal=11
Malloc Size=131160 LargeMemoryPoolOffset=393352
Malloc Size=131160 LargeMemoryPoolOffset=524536
Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
Segmentation fault (core dumped)
```

I understand that the minimum recommended is 3070, but that has 8GB, same as my 2070 Super.
Are there other requirements that RTX2070 Super does not qualify for to run UE5?

Thank you

closed 2025-01-14 by MarcelPiNacy-CVC

AxoyTO, 2025-05-15:
My RTX 3080Ti is crashing 10 secs after moving around the map. Only the launch takes around 8G in my GPU Memory...