carla-simulator/carla #8529: Fatal Error When Spawning Vehicle with Semantic Lidar in Mine_01

VitaminQRQ, 2025-01-07:
CARLA version: 0.10.0  

Platform/OS: Windows 11 and Ubuntu 22.04
  
Problem you have experienced: 
Unable to spawn a vehicle with semantic lidar in Mine_01. When I add a semantic lidar sensor, CARLA encounters a fatal error and exits. The same issue occurs on both Windows and Ubuntu. However, the same sensor configuration works fine in Town10.  

What you expected to happen:  
Successfully spawn a vehicle with a semantic lidar sensor.  

Steps to reproduce:  
Replace the `stack.json` file in the ROS2 example with the attached content and run `ros2_native.py`:

```json
{
    "type": "vehicle.lincoln.mkz",
    "id": "ego",
    "sensors": [
        {
            "type": "sensor.camera.rgb",
            "id": "rgb",
            "spawn_point": {"x": -4.5, "y": 0.0, "z": 2.5, "roll": 0.0, "pitch": 20.0, "yaw": 0.0},
            "attributes": {
                "image_size_x": 400,
                "image_size_y": 200,
                "fov": 90.0
            }
        },
        {
            "type": "sensor.lidar.ray_cast",
            "id": "lidar",
            "spawn_point": {"x": 0.0, "y": 0.0, "z": 2.6, "roll": 0.0, "pitch": 0.0, "yaw": 0.0},
            "attributes": {
                "range": 100,
                "channels": 64,
                "points_per_second": 600000,
                "rotation_frequency": 20,
                "upper_fov": 10,
                "lower_fov": -30,
                "atmosphere_attenuation_rate": 0.004,
                "dropoff_general_rate": 0.45,
                "dropoff_intensity_limit": 0.8,
                "dropoff_zero_intensity": 0.4
            }
        },
        {
            "type": "sensor.lidar.ray_cast_semantic",
            "id": "lidar_semantic",
            "spawn_point": {"x": 0.0, "y": 0.0, "z": 2.6, "roll": 0.0, "pitch": 0.0, "yaw": 0.0},
            "attributes": {
                "range": 85,
                "channels": 64,
                "points_per_second": 600000,
                "rotation_frequency": 20,
                "upper_fov": 10,
                "lower_fov": -30
            }
        },
        {
            "type": "sensor.other.gnss",
            "id": "gnss",
            "spawn_point": {"x": 1.0, "y": 0.0, "z": 2.0, "roll": 0.0, "pitch": 0.0, "yaw": 0.0},
            "attributes": {
                "noise_alt_stddev": 0.0, 
                "noise_lat_stddev": 0.0, 
                "noise_lon_stddev": 0.0,
                "noise_alt_bias": 0.0, 
                "noise_lat_bias": 0.0, 
                "noise_lon_bias": 0.0
            }
        },
        {
            "type": "sensor.other.imu",
            "id": "imu",
            "spawn_point": {"x": 2.0, "y": 0.0, "z": 2.0, "roll": 0.0, "pitch": 0.0, "yaw": 0.0},
            "attributes": {
                "noise_accel_stddev_x": 0.0, 
                "noise_accel_stddev_y": 0.0, 
                "noise_accel_stddev_z": 0.0,
                "noise_gyro_stddev_x": 0.0, 
                "noise_gyro_stddev_y": 0.0, 
                "noise_gyro_stddev_z": 0.0,
                "noise_gyro_bias_x": 0.0, 
                "noise_gyro_bias_y": 0.0, 
                "noise_gyro_bias_z": 0.0
            }
        }
    ]
}

joel-mb, 2025-01-08:
@Blyron @VitaminQRQ I was able to reproduce the error.

It seems to be related to the `Mine_01` map, rather than the ROS2 integration.

The error can also be reproduced by using manual control and simply changing the sensor until the semantic lidar gets spawned.

@jorge-kabuto Could you please take a look?

PabloVD, 2025-01-10:
I have run the manual control in the mine in the editor, to gather more information. When switching to semantic lidar, it also crashes. Here is the error log, which confirms that the problem is related to the semantic lidar rather than ros2:

LoginId:cdb5cd8a3e3849138ff555c3fccaaf81-000003e8

Assertion failed: (Index >= 0) & (Index < ArrayNum) [File:Runtime/Core/Public/Containers/Array.h] [Line: 783] 
Array index out of bounds: 0 into an array of size 0

libUnrealEditor-Carla.so!ARayCastSemanticLidar::ComputeRawDetection(FHitResult const&, UE::Math::TTransform<double> const&, carla::sensor::data::SemanticLidarDetection&) const [/home/tda/CARLA/CARLAbuild/UnrealEngine/Engine/Source/Runtime/Core/Public/Containers/Array.h:783]
libUnrealEditor-Carla.so!ARayCastSemanticLidar::ComputeAndSaveDetections(UE::Math::TTransform<double> const&) [/home/tda/CARLA/CARLAbuild/carlaUE5/Unreal/CarlaUnreal/Plugins/Carla/Source/Carla/Sensor/RayCastSemanticLidar.cpp:200]
libUnrealEditor-Carla.so!ARayCastSemanticLidar::SimulateLidar(float) [/home/tda/CARLA/CARLAbuild/carlaUE5/Unreal/CarlaUnreal/Plugins/Carla/Source/Carla/Sensor/RayCastSemanticLidar.cpp:159]
libUnrealEditor-Carla.so!ARayCastSemanticLidar::PostPhysTick(UWorld*, ELevelTick, float) [/home/tda/CARLA/CARLAbuild/carlaUE5/Unreal/CarlaUnreal/Plugins/Carla/Source/Carla/Sensor/RayCastSemanticLidar.cpp:75]
libUnrealEditor-Carla.so!ASensor::PostPhysTickInternal(UWorld*, ELevelTick, float) [/home/tda/CARLA/CARLAbuild/carlaUE5/Unreal/CarlaUnreal/Plugins/Carla/Source/Carla/Sensor/Sensor.cpp:136]
libUnrealEditor-Carla.so!FSensorManager::PostPhysTick(UWorld*, ELevelTick, float) [/home/tda/CARLA/CARLAbuild/carlaUE5/Unreal/CarlaUnreal/Plugins/Carla/Source/Carla/Sensor/SensorManager.cpp:25]
libUnrealEditor-Carla.so!FCarlaEngine::OnPostTick(UWorld*, ELevelTick, float) [/home/tda/CARLA/CARLAbuild/carlaUE5/Unreal/CarlaUnreal/Plugins/Carla/Source/Carla/Game/CarlaEngine.cpp:383]
libUnrealEditor-Carla.so!TBaseRawMethodDelegateInstance<false, FCarlaEngine, void (UWorld*, ELevelTick, float), FDefaultDelegateUserPolicy>::ExecuteIfSafe(UWorld*, ELevelTick, float) const [/home/tda/CARLA/CARLAbuild/UnrealEngine/Engine/Source/Runtime/Core/Public/Templates/Invoke.h:66]
libUnrealEditor-Engine.so!void TMulticastDelegateBase<FDefaultDelegateUserPolicy>::Broadcast<IBaseDelegateInstance<void (UWorld*, ELevelTick, float), FDefaultDelegateUserPolicy>, UWorld*, ELevelTick, float>(UWorld*, ELevelTick, float) const [/home/tda/CARLA/CARLAbuild/UnrealEngine/Engine/Source/Runtime/Core/Public/Delegates/MulticastDelegateBase.h:257]
libUnrealEditor-Engine.so!UWorld::Tick(ELevelTick, float) [/home/tda/CARLA/CARLAbuild/UnrealEngine/Engine/Source/Runtime/Core/Public/Delegates/DelegateSignatureImpl.inl:1079]
libUnrealEditor-UnrealEd.so!UEditorEngine::Tick(float, bool) [/home/tda/CARLA/CARLAbuild/UnrealEngine/Engine/Source/./Editor/UnrealEd/Private/EditorEngine.cpp:2138]
libUnrealEditor-UnrealEd.so!UUnrealEdEngine::Tick(float, bool) [/home/tda/CARLA/CARLAbuild/UnrealEngine/Engine/Source/./Editor/UnrealEd/Private/UnrealEdEngine.cpp:547]
UnrealEditor!FEngineLoop::Tick() [/home/tda/CARLA/CARLAbuild/UnrealEngine/Engine/Source/./Runtime/Launch/Private/LaunchEngineLoop.cpp:5863]
UnrealEditor!GuardedMain(char16_t const*) [/home/tda/CARLA/CARLAbuild/UnrealEngine/Engine/Source/./Runtime/Launch/Private/Launch.cpp:69]
libUnrealEditor-UnixCommonStartup.so!CommonUnixMain(int, char**, int (*)(char16_t const*), void (*)()) [/home/tda/CARLA/CARLAbuild/UnrealEngine/Engine/Source/./Runtime/Unix/UnixCommonStartup/Private/UnixCommonStartup.cpp:323]
libc.so.6!__libc_start_main(+0xf2)
UnrealEditor!_start(+0x29)

closed 2025-01-16 by PabloVD