carla-simulator/carla #8536: Custom map package exported from windows build version cannot work on ubuntu package installed version

sleepyLeo6, 2025-01-08:
CARLA version: 0.9.15
Platform/OS: Windows11 build version, Ubuntu 20.04 package installed version

I've used RoadRunner to generate a custom map and imported it to my windows build version Carla successfully. Then I wanted to use the custom map on package installed version Carla. 

So I followed [Create distribution packages for assets](https://carla.readthedocs.io/en/0.9.15/tuto_A_create_standalone/#export-a-package-in-a-carla-build-from-source) to export the custom map as a standalone package. 

Firstly I tried to import the custom map package to a windows package installed version Carla. I uziped the pacakge zip file at the `CARLA_ROOT\WindowsNoEditor\` path. It seems nothing wrong with this. I can switch to the custom map in carla simulation using python scripts.

Then I tried this on a Ubuntu package installed version. I moved the map package to the `Import` folder and run `./ImportAssets.sh` at carla root path. When I tried to switch to the custom map, it worked at first, the map can be loaded in Carla simulation. But after a while, the Carla crashed with faults like below:
```
4.26.2-0+++UE4+Release-4.26 522 0
Disabling core dumps.
WARNING: cannot parse georeference: ''. Using default values. 
Signal 11 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554 
CommonUnixCrashHandler: Signal=11
Malloc Size=131160 LargeMemoryPoolOffset=196744 
Malloc Size=131160 LargeMemoryPoolOffset=327928 
Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
Segmentation fault (core dumped)
``` 

Why would this happen? And is there any possibility that I can import a custom map package generated from Windows build version to a Ubuntu package installed version?

AliSafaeifar, 2025-01-08:
Hi,

I’ve encountered the same issue before. While I can’t recall all the exact details, I eventually resolved it by installing the "build from source" version of Carla on Ubuntu as well (as other methods didn’t work). First, I prepare the map using the Windows "build from source" version (primarily because I found it more convenient and had RoadRunner installed on Windows). Then, I copy the map files (located in the Content folder) to the same directory in Linux. From there, I use the "build from source" version of Carla on Ubuntu to create the packaged version of the map.

One important note: the "build from source" version of Carla on Ubuntu must match the version of the binary Carla that you’ll later use with the packaged map. If the versions don’t align, you may run into issues, particularly with shadows and other graphical elements, when using the map

sleepyLeo6, 2025-01-08:
> Hi,
>
> I’ve encountered the same issue before. While I can’t recall all the exact details, I eventually resolved it by installing the "build from source" version of Carla on Ubuntu as well (as other [...]

Thanks! I’m still curious if there’s a way to directly use the map package exported from the Windows source build. But I’ll try your suggested solution.