carla-simulator/carla #8537: Signal 11 caught

zdy1013, 2025-01-08:
I used Carla 0.9.15 to run Carla Leaderboard 2.0, and opened 3 Carla servers in parallel. They all worked normally at the beginning, but after running for a long time, this problem occurred:

4.26.2-0+++UE4+Release-4.26 522 0
Disabling core dumps.
Signal 11 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554
CommonUnixCrashHandler: Signal=11
Malloc Size=131160 LargeMemoryPoolOffset=196744
Malloc Size=131160 LargeMemoryPoolOffset=327928
Engine crash handling finished;  re-raising signal 11 for the default handler. Good bye.
Segmentation fault (core dumped)

My graphics card is 2*RTX3090 graphics card driver is 535.161.07, cuda version is 12.0, memory is 64GB, torch version is 2.2.0, vulkan version is 1.3.280.0, and vulkaninfo output is normal. And then I'm going to start with opengl, and I'm going to do the same thing over time. This is my startup command:
# Start the first CARLA server instance
./CarlaUE4.sh -RenderOffScreen -nosound -carla-rpc-port=30000 -graphicsadapter=0 -opengl
./CarlaUE4.sh -RenderOffScreen -nosound -carla-rpc-port=35432 -graphicsadapter=0 -opengl
./CarlaUE4.sh -RenderOffScreen -nosound -carla-rpc-port=43234 -graphicsadapter=1 -opengl

Please give me your advice. Thank you

kostasoiko4, 2025-01-22:
I face the same issue, trying to run a PPO algorythm to train a self-driving car agent. The simulator run for anywhere between 15 minutes and 1 hour and then i have the same crash message. It also runs on an Ubuntu system with double RTX 3080s. Couldn't find any specific solution other than "Update the gpu drivers". In my experience the "heavier" the script is the more often i see this crash report, i run the same script for a training senario with a single front facing camera, with 5 cameras and they both managed to reach 500K episodes to be good enough for a baseline example. Now i am running exactly the same scritp with 5 cameras, 5 radars and a lidar and i can't get more than 70K-80K episodes. 

The simulator along side the training script most of the time take about 6-7GB of GPU memory. Is it really possible that the simulator just spikes up and takes all the available memory from the system and that's the crash reason?

potatoler, 2025-08-13:
Same issue on 0.9.15 with Vulkan. The simulator takes up about 8G of vRAM and logs exactly same as mentioned above：
```
4.26.2-0+++UE4+Release-4.26 522 0
Disabling core dumps.
Signal 11 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554
CommonUnixCrashHandler: Signal=11
Malloc Size=131160 LargeMemoryPoolOffset=196744
Malloc Size=131160 LargeMemoryPoolOffset=327928
Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
Segmentation fault (core dumped)
```

Rak-r, 2025-08-13:
> Same issue on 0.9.15 with Vulkan. The simulator takes up about 8G of vRAM and logs exactly same as mentioned above：
>
> ```
> 4.26.2-0+++UE4+Release-4.26 522 0
> Disabling core dumps.
> Signal 11 caught. [...]

I am having the same issue. For more details please see [https://github.com/carla-simulator/carla/issues/9152#issue-3314232492](url) . 

I also have an underlying issue with Vulkan, due to which the server gets crashed even with the default scripts of pythonapi tutorial.

orjahren, 2025-11-26:
> My graphics card is 2*RTX3090 graphics card driver is 535.161.07

According to https://github.com/Thinklab-SJTU/Bench2Drive#deal-with-carla : 

> We find that nvidia driver version 470 is good all the time. 515 has some problems but okay. 550 has lots of bugs.

So, maybe testing an older driver could be relevant? That project also uses version 0.9.15.