carla-simulator/carla #8540: build carlaue5 error

uuauto2024, 2025-01-08:
I encountered a problem while compiling the latest version of carlaue5. Specifically, the issue occurred during the 
“make”
（
echo "Setup CARLA UnrealEngine5..."
./Setup.sh --force
echo "GenerateProjectFiles CARLA UnrealEngine5..."
./GenerateProjectFiles.sh
echo "Build CARLA UnrealEngine5..."
**make**
popd
popd
）
 step in building the Unreal Engine. The error is as follows, please help me, thank you.

----------------------------------------------------------------------------------------------------------------------
[Adaptive Build] Excluded from VirtualCamera unity file: VirtualCamera.cpp, VCamUserSettings.cpp, AssetFilteringAndSortingFunctionLibrary.cpp, LevelSequenceVCamLibrary.cpp, MultiUserTakesVCamFunctionLibrary.cpp, VCamBlueprintFunctionLibrary.cpp, TakeMetaDataTagsFunctionLibrary.cpp, GameViewFunctionLibrary.cpp, VirtualCameraClipsMetaData.cpp
[Adaptive Build] Excluded from VirtualCameraEditor unity file: VirtualCameraEditorModule.cpp
[Adaptive Build] Excluded from PixelStreamingVCam unity file: VCamPixelStreamingSessionLogic.cpp, PixelStreamingVCamLog.cpp, VCamPixelStreamingSubsystem.cpp, PixelStreamingVCamModule.cpp, SignalingServerLifecycle.cpp, VirtualCameraBeaconReceiver.cpp, VCamPixelStreamingLiveLink.cpp, LiveLinkManager.cpp, PixelStreamingMediaOutput.cpp, PixelStreamingVideoInputVCam.cpp, MissingSignallingServerNotifier.cpp
[Adaptive Build] Excluded from VCamBlueprintNodes unity file: VCamBlueprintNodes.cpp, VCamBlueprintNodesLog.cpp, VCamK2Node_SwitchConnectionSystemBase.cpp, VCamK2Node_SwitchOnWidgetConnections.cpp, VCamK2Node_SwitchOnModifierConnectionPoints.cpp
[Adaptive Build] Excluded from HttpBlueprint unity file: HttpHeader.cpp, HttpBlueprintFunctionLibrary.cpp, HttpBlueprintModule.cpp, HttpRequestProxyObject.cpp
[Adaptive Build] Excluded from HttpBlueprintGraph unity file: K2Node_HttpRequest.cpp, K2Node_AsyncMakeRequestHeader.cpp, HttpBlueprintGraphLog.cpp, HttpBlueprintGraphModule.cpp, K2Node_MakeRequestHeader.cpp
Determining max actions to execute in parallel (40 physical cores, 80 logical cores)
  Executing up to 40 processes, one per physical core
Using Unreal Build Accelerator local executor to run 4 action(s)
  Storage capacity 40Gb
---- Starting trace ----
UbaSessionServer - Disable remote execution (remote sessions will finish current processes)
------ Building 4 action(s) started ------
[1/4] Link (lld) libUnrealEditor-UserToolBoxCore.so (UBA disabled)
ld.lld: error: undefined symbol: UUserToolBoxBaseBlueprint::Execute()
>>> referenced by Module.UserToolBoxCore.cpp
>>>               ../Plugins/Experimental/UserToolBoxCore/Intermediate/Build/Linux/x64/UnrealEditor/Development/UserToolBoxCore/Module.UserToolBoxCore.cpp.o:(vtable for UUserToolBoxBaseBlueprint)
clang++: error: linker command failed with exit code 1 (use -v to see invocation)
Trace file written to /home/simulator/UnrealEngine5_carla/Engine/Programs/UnrealBuildTool/Log.uba with size 1.7kb
Total time in Unreal Build Accelerator local executor: 0.55 seconds
Total execution time: 151.30 seconds

----------------------------------------------------------------------------------------------------------------------

Blyron, 2025-01-08:
Which System are you using?

uuauto2024, 2025-01-08:
> Which System are you using?

Ubuntu 22.04.5

MarcelPiNacy-CVC, 2025-01-08:
Could you share the entire log? Ideally attach it as a separate file, otherwise formatting breaks.

uuauto2024, 2025-01-09:
> Could you share the entire log? Ideally attach it as a separate file, otherwise formatting breaks.

Here is the entire lof file, thank you.
https://drive.google.com/file/d/13iwmXAV4NiQgYmiBrf2Uk_-SXDmon-Nk/view?usp=drive_link

uuauto2024, 2025-01-09:
> Could you share the entire log? Ideally attach it as a separate file, otherwise formatting breaks.

Here is the entire log file, thank you.

https://drive.google.com/file/d/13iwmXAV4NiQgYmiBrf2Uk_-SXDmon-Nk/view?usp=drive_link

MarcelPiNacy-CVC, 2025-01-15:
You could try to build UE5 separately and then try to open CARLA afterwards.
Try running `make` in the UE5 folder for now. Then directly use CMake commands to build CARLA.

closed 2025-01-27 by MarcelPiNacy-CVC

by1e11, 2025-03-11:
Hi guys, is there any solution for this? I encountered the exactly same issue